mavlink/qgroundcontrol #5264: Large video lag after moving from 3.1.3 [Windows only]

jaxxzer, 2017-06-08:
I am in touch with two users who have machines exhibiting the same behavior. One is on Windows 7 and one is on Windows 10. They have nice low latency streams on 3.1.3, but a great deal of lag (1-2 s) on builds from the last month or so. I have tried to reproduce this on four machines with the same camera hardware and software (raspberry pi camera) as the users, to no avail.

Here are some details from the Windows 10 user:

[image]
[image]

https://drive.google.com/file/d/0B5HoJwmhA7VKUVY4NVpJYjRWQ0NyVVJhU0Rzczhhc1BwZkZR/view?usp=sharing

DonLakeFlyer, 2017-06-09:
I would only guess that it's some sort of gstreamer+video driver problem that we don't have control over.

BoKoppel, 2017-06-09:
I have the Win 7 machine.
Spec of PC is i7, 2,8GHz, 8Gb mem, 64bit OS, Swedish OS (Home premium)
I also run a slower (i3) PC with English Win7 PRO, that machine has no latency.
Since both problem PCs has scandinavian OS, is there a hint?
When I play the stream direct from Gstreamer it is no lag(Latency)

NaterGator, 2017-06-09:
Wow. Off the bat that does not look like a network error. That is either a clocking problem on the gstreamer pipeline or an incorrect frame rate in negotiated caps. 

Can we have them run the application with the environment variable GST_DEBUG set to either 4 or 5 (5 is going to produce a lot of debugging output) and capture stderr?

jaxxzer, 2017-06-09:
Thanks @NaterGator, I'll walk them through that.

NaterGator, 2017-06-09:
We should perhaps consider setting sync=false on the qtquick2videosink. We're not really interested in pipeline sync'ed presentation right now, we just want to present frames to the user as quickly as they come in.

jaxxzer, 2017-06-10:
> We should perhaps consider setting sync=false on the qtquick2videosink. 

@NaterGator unless I'm missing something, this is already the case in the offending builds: https://github.com/mavlink/qgroundcontrol/commit/41771effe76fc50aca6952cf93ee3f235257c11b

jaxxzer, 2017-06-13:
The list of people I've heard from about this has grown to 8  or 10. It looks like it might have something to do with https://github.com/mavlink/qgroundcontrol/commit/ce3cba6e5603471e81a9a2cdedcbe8460dad8255 (reverting that fixed the issue for some), but I'm still not sure at this point.

I'm trying to build QGC with GST_DEBUG output to a file, so these people don't have to go to a command line themselves. I've modified [this bit](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/VideoStreaming.cc#L123) to write to a file, but it doens't work with on windows due to the <unistd.h> requirement. @NaterGator @dogmaphobic can you help me with this?

BoKoppel, 2017-07-01:
I have no problem doing logging through command window if that helps tracking problem.
I need syntax in that case.

dogmaphobic, 2017-07-01:
For people with an actual Windows machine and able to build/debug, try using a newer version of GStreamer than 1.5.2. The only reason we settled on that version was because it was working everywhere and changing version wholesale is a lot of work (CI). Given that we've been on the same version for around 2 years now, it may be time to explore a newer version.

DonLakeFlyer, 2017-07-01:
@antiheavy Have you guys been running video?

Antiheavy, 2017-07-01:
yes, we've been running video, although not much recently.  We noticed the video performance seemed to change around the time of 3.1.3 as well.  We looked into why a little, but I don't know that we got to the bottom of it.  We did find that our older Window's laptops could no longer keep up with the video, even after updating drivers.  Newer, faster laptops seem to be able to keep up with the video.  That's the extent of what I know currently.  We plan to start testing video performance more extensively in the coming weeks so I'm happy to test any improvements people come up with.

DonLakeFlyer, 2017-07-02:
Is it possible to do a diff between the important bits of video in 3.1.3 to master to see what important is different?

BoKoppel, 2017-07-02:
Problem Went up when QGC versions introduced video recording.
The problem is the same with or without video recording used.
Latency around 2 seconds, and a LOT of framedrop.
Both in live feed as well as recording.
Cutting down to 10Mbit and 12 fps from the camera make latency 200msec again, but not good Quality.
Latest QGC versions without recording made things good again.
Older Intel i3 PC,s have not problem, newer i7 Machines have.
Problem is not processor load or network load.
Probably somthing with Gstream pipeline.
I have triedn installing newer G stream, same problem, BUT newer Gstream do not install under QGC folder, so I guess QGC still use older verison.
I have filmed the screen via iPhone, and made QGC recording, link to result:
https://www.dropbox.com/sh/hqqh03p721qlixp/AACoPAy48xU7R-cl0TmVZzEta?dl=0

DonLakeFlyer, 2017-07-02:
Ok, thanks. I'd really like to get 3.2 out to door. So if this is going to be hard to figure out then my suggestion is to remove video recording support. Non-laggy video seems way more important at this point.

BoKoppel, 2017-07-02:
I agree regarding priority
I'm happy if I can contribute with latency testing, logging etc.
I'm not in the compiling world, Command prompt and .exe testversions no problems.....

DonLakeFlyer, 2017-07-02:
I'll probably cut first 3.2 build in about a week. So if you guys can figure anything out before then great. @dogmaphobic isn't much of a windows person, and I know nothing about the video stuff. So maybe @NaterGator is one the few people that can help out here.

dogmaphobic, 2017-07-02:
If this is a Windows only issue, #ifdef it out of Windows. It doesn't make sense to remove the functionality altogether given that it works fine everywhere else.

dogmaphobic, 2017-07-02:
Even a dumb, low end Android can handle this. It ought to be something peculiar about the Windows implementation.

DonLakeFlyer, 2017-07-02:
Windows problems have a tendency to have something to do with the OpenGL drivers.

BoKoppel, 2017-07-03:
OpenGL drivers; since problem occure also at recording, do QGC realy use OpenGL for recording?

BoKoppel, 2017-07-03:
More info that might help pinpoint development:
I tested video via gstreamer 1.5.2 from Command window, no QGC running:
gst-launch-1.0 udpsrc port=5600 ! application/x-rtp, encoding-name=H264, payload=96 ! rtph264depay ! h264parse ! avdec_h264 ! autovideosink
Result: OK latency (200mSec),  minor framedrops

Uninstalled Gstreamer, still 2 sec latency in QGC versions with recording (2017-04-23 and 2017-06-06)
Seems like QGC only use its own installed gstreamer files in the QGC folder, not the Gstreamer files installed in c:\gstreamer

Tested latest QGC version 1.1.4.5
OK Latency (200mSec), minor framedrops, looked like using Gstreamer direct

-Latency 2 sec problem seems related to QGC with recording option settings in gstreamer.
-Minor framedrops might always be the result in all setups? You probably not notice this without test.
Videos from tests in Dropbox (link in earlier post in same thread)
Is there more tests I can do?

jaxxzer, 2017-07-03:
1.1.4.5 that @BoKoppel is referring to is my own build (in releases on my fork). I made several builds reverting different changes to the video code since 3.1.3 in order to try to narrow down the problem with those experiencing the issue. After the feedback I have received from those who tested I am suspecting the issue stems from https://github.com/mavlink/qgroundcontrol/commit/ce3cba6e5603471e81a9a2cdedcbe8460dad8255, but I'm not sure.

BoKoppel, 2017-08-03:
We have a solution for our computers:
[](http://discuss.bluerobotics.com/t/video-delay-in-qgc/1335url)

DonLakeFlyer, 2017-08-03:
Ok, that's a bizarre one :)

closed 2017-08-03 by DonLakeFlyer