mavlink/qgroundcontrol #5273: qgroundcontrol android does not take a wakelock and allows the app to go to sleep and the screen to turn off

marcmerlin, 2017-06-09:
I used it on the field, it worked fine as long as I kept the phone awake through other means.
Sadly I cannot turn off the screen saver lock because it's required by my phone policy.
That said, tower allows me to turn off the screen while tower keeps running and sends me audio warnings during my flight as well as records said flight. You may ask @billbonney what code he has in Tower to allow this, or just look in his tree :)
qgroundcontrol should obviously allow the same. Running in the background with the screen off is actually an important feature to save the phone batteries
For now I use a 3rd party app that takes a full wakelock and burn batteries but at least allows qgroundcontrol to keep running without going to sleep.

DonLakeFlyer, 2017-06-10:
Given the library we use to build QGC I don't think it is possible for it to run when locked. Tower and QGC are very different apps, Tower is a native android java app, whereas QGC is a C++ app.

marcmerlin, 2017-06-10:
I'm not an android expert, but does this help?
https://stackoverflow.com/questions/17654140/is-there-a-way-to-get-wakelock-on-android-through-the-jni-ndk
Otherwise, telling the user with a popup to install a 3rd party wakelock app and what wakelock setting to use, would be helpful.

DonLakeFlyer, 2017-06-10:
> That said, tower allows me to turn off the screen while tower keeps running and sends me audio warnings during my flight as well as records said flight.

This is because part of Tower runs as an Android service which is not possible with QGC.

closed 2026-10-05 by github-actions[bot]