mavlink/qgroundcontrol #5276: android/qgroundcontrol should auto record tlog in a directory under /sdcard

marcmerlin, 2017-06-10:
I believe qgroundcontrol has a way to manually save tlogs, but it should do so automatically with no user input.
I did a few flights with qgroundcontrol on android yesterday, and none of those flights got recorded on my phone. 
Most of the time, sadly qgroundcontrol stopped running due to the phone going to sleep, or some other reason, so the main point is that it should automatically record all flights without user input until the app dies or gets terminated (which can be done at inconvenient times on android vs a desktop version).
While I do like qgroundcontrol's interface and extra functionality, for now I have to go back to Tower for the reliable auto track recording feature.
But I'm looking forward to going back to qgroundcontrol as soon as the track recording is reliable :)

DonLakeFlyer, 2017-06-11:
@dogmaphobic I thought you had tuned on support for telemetry logs on Android but now on by default. But it look like the settings doesn't show up.

dogmaphobic, 2017-06-13:
It "should" be there. I'll take a look when I get back.

dogmaphobic, 2017-06-27:
I spent some time on this today and I can't quite tell what's going on. The _Start Log_ is properly sent and acknowledged, the log file is created but no log data is ever received. I'm using SITL as I don't have anything functional at the moment. After arming/disarming the "vehicle" several times, I ended up with several log files but obviously all with zero length as no data ever arrived.

DonLakeFlyer, 2017-06-27:
If the setting is showing up and I can turn it on then I can debug from there.

dogmaphobic, 2017-06-27:
It's all there. The same thing is happening on desktop. No log data arrives.

dogmaphobic, 2017-06-27:
Does SITL support this?

DonLakeFlyer, 2017-06-27:
Sure. Telemetry logs are just mavlink traffic.

dogmaphobic, 2017-06-27:
No, this is not telemetry log. These are ulog logs (as described above).

dogmaphobic, 2017-06-27:
>I have "enable automatic logging" in Mavlink 2.0 Logging (PX4 Firmware Only)

DonLakeFlyer, 2017-06-28:
@marcmerlin I'm confused: You talk about saving tlogs which are telemetry logs. But you also talk about using "enable automatic logging" in Mavlink 2.0 Logging (PX4 Firmware Only)" which are vehicle logs, which are not telemetry logs. Do you want telemetry logs on mobile?

@dogmaphobic I was talking about saving telemetry logs on mobile which I though you had made available on mobile builds but I don't see the options showing up for that when I build android master.

@dogmaphobic With regards to the vehicle log automomatic logging stuff: Are you running that over a mavlink 2 connection? If you aren't I assume it will not work. Does the code check that it has a mavlink 2 connection before it tries to do anything? Also I would suggest changing the UI for this to better indicate that these are vehicle .ulogs (whatever the name for those are) and no telemetry logs.

DonLakeFlyer, 2017-06-28:
I think we also need to look at app sleep again: https://github.com/mavlink/qgroundcontrol/issues/5354

dogmaphobic, 2017-06-28:
Ok, this is as far as I got with telemetry log on Android (mobile in general). If logging is enabled, it is created and written to. Log is saved to a temp file and at the end, it is copied to the final location. On mobile, this never happens because the call that does all that is never called. Here is the function that closes (and eventually issues the signal to do the copy):

https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L330-L336

I disconnecting the vehicle (shutting down SITL). It took QGC about 20 seconds to figure out communication was lost. I proceeded to "Disconnect". In that case the log was properly saved.

In summary, logging is being written but never finalized unless you disconnect from the vehicle first. My guess is the application shutdown order on mobile is different and `_vehicleCountChanged()` is never called on exit.

Note that all this was only possible after forcibly enabling telemetry log, which is disabled by default on mobile and there is no UI to enable it.

[image]

marcmerlin, 2017-06-28:
@DonLakeFlyer I want/need to log tlog on my phone. 
I think you just explained to me that what I tried to turn on does not do what I actually need.
A GCS should have a way to log tlog for debugging or vehicle recovery if you have a crash in a remote location, ideally it should do so automatically (the other GCS I tried, do). 
Does qgroundcontrol allow this?

DonLakeFlyer, 2017-06-28:
That seems like it is working correctly. The log is written when:
* Vehicle disconnects
* Or last vehicle disarms (depending on settings)

Since mobile has strange shutdown behavior you need to do one of the above to get it to close out. I think that is ok.

DonLakeFlyer, 2017-06-28:
Given the mobile shutdown behavior the options to save logs even if vehicle is not armed may not be a good one. If you have everything else working could you put it up in a pull then I'll work with that to figure out what to do.

DonLakeFlyer, 2017-06-28:
> Does qgroundcontrol allow this?

@marcmerlin Working on it.

marcmerlin, 2017-06-28:
@DonLakeFlyer what you describe on writing the log (disconnect or disarm) sounds great, it's just not what I saw when I tried it.
Actually I got no log file at all of any of the flights I did.
Just to be clear, is it supposed to auto start saving of tlog under /sdcard somewhere? If so, where?
If it's not automatic, what is the user supposed to do to make it automatic?
Or your last update is saying that it's not doing this yet, but that you are working on it so that it will?

DonLakeFlyer, 2017-06-28:
You couldn't have tried it since you can't currently turn it on from android. What you used was to download binary logs from the vehicle. That is a different thing.

marcmerlin, 2017-06-28:
@DonLakeFlyer ok, that clears up the confusion, thanks.
I'll wait for a new version where this is turned on, thanks for your replies and looking at this.

DonLakeFlyer, 2017-07-01:
Definitely something wrong with shutdown sequence on android

DonLakeFlyer, 2017-07-01:
All working now. @marcmerlin Can you retry with tomorrow's android daily and let us know how it worked for you? Should be fully through the build system by then.

closed 2017-07-03 by DonLakeFlyer

marcmerlin, 2017-07-04:
@DonLakeFlyer thanks for doing this. Sadly, all 3 of my RC aircraft got destroyed last week (2 yesterday), so right now I'm unable to test anything :(

DonLakeFlyer, 2017-07-06:
Yikes!