mavlink/qgroundcontrol #5282: ArduPIlot: Pre-Flight baro help text update

DonLakeFlyer, 2017-06-12:
From @wickedshell: "I know it's really annoying and verbose but for the plane one I would consider adding a note that trys to explain that you want to block the tube from the wind without sealing any of the holes on the probe"

Need to come up with something better.

DonLakeFlyer, 2017-06-12:
For docs: "To calibrate the airspeed sensor shield it from the wind, without blocking any of the holes on the airspeed sensor."

DonLakeFlyer, 2017-06-12:
PX4 text should change as well to be better.

hamishwillee, 2017-06-12:
How about: "You must not touch the sensor or obstruct any holes during the calibration.' OR "Do not touch the sensor or obstruct any holes during the calibration."

closed 2017-06-20 by DonLakeFlyer