mavlink/qgroundcontrol #5284: MAVLink Inspector shows no packets received if stream does no include heartbeat

magicrub, 2017-06-12:
When using the MAVLink Inspector, zero packets show up if you're not also sending a heartbeat in the stream. Shouldn't the inspector show everything no matter what, with and without a heartbeat?

DonLakeFlyer, 2017-06-12:
The mavlink inspector is keyed to vehicles as opposed to communication streams. A heartbeat is required to crank up a vehicle. So if you have no vehicle, then you won't get anything. Seems like an odd case where there is not heartbeat?

magicrub, 2017-06-12:
well, I was using QGC as a debug too to analyze packets but nothing was showing up. Didn't realize it dropped incoming packets if no heartbeat was found. I would expect there to be a "unknown vehicle" section or equivalent.

DonLakeFlyer, 2017-12-30:
Going to leave this alone

closed 2017-12-30 by DonLakeFlyer