mavlink/qgroundcontrol #5294: Toolbar is too large in daily build.

sh83, 2017-06-14:
Main toolbar buttons consume too much space on phone screen. Appears only in daily build. Latest release is ok. See screenshots for detail.
[image]
[image]

DonLakeFlyer, 2017-06-14:
What type/size device?

sh83, 2017-06-14:
It is andrid smartphone: oneplus one, 1920×1080 px, 5.5'.

alpha-numericon, 2017-06-20:
Same issue: 1920x1080, 4.95" Android phone. Looks like it was introduced in 1f77b2f7113891bf395c07d410607f3068531db5.

sh83, 2017-06-28:
Fixed in last build. Thank you.

closed 2017-06-28 by sh83

DonLakeFlyer, 2017-06-28:
Great, thanks for verifying.