mavlink/qgroundcontrol #5295: QGC 3.1.3: bounding box shown for offline map tile set is incorrect

tstellanova, 2017-06-14:
Here's how I encountered this:
- add a new tile set in Offline Maps
- select an area less than a mile wide in each dimension
- select Bing Maps Hybrid
- click Download
- Return to list of offline tile sets
- select the tile set you just created
- a map of the entire world is presented. If you zoom in on the region you selected for the tile site, a long narrow, half-world-wide red bounding box is presented as the bounding box of your tile set.
- zoom in further on the tile set region, and the red bounding box extends off to (Japan) from your actual tile set region.  A maybe 10% alpha red overlay indicates your actual tile set region.
- See attached image.
- Tested on Mac OSX 10.11.16

[image]

dogmaphobic, 2017-06-27:
I repeated the steps to the letter. Twice. Could not reproduce it.

[image]

DonLakeFlyer, 2017-06-27:
@tstellanova Can you try to reproduce your steps using a daily build?

dogmaphobic, 2017-06-27:
Bing "Hybrid" (not that it would make any difference)
Also, I'm running maxOS 10.12.5 but I can't think of a reason for the to be an issue either.

[image]

NaterGator, 2017-06-27:
I've reproduced this once or twice. I'll see if I can get a consistent
repro on Windows.

DonLakeFlyer, 2017-06-29:
This smells like a Qml MapQuickItem bug. This sort of thing happens with mission planner Survey items every once in a while as well. I've never been able to nail it down.

DonLakeFlyer, 2017-07-07:
Does anyone have a consistent repro case?

DonLakeFlyer, 2017-12-30:
Without a reproducible case not much we can do here

closed 2017-12-30 by DonLakeFlyer