mavlink/qgroundcontrol #5298: Not able to set SENS_EN_LL40LS to 2

ChristophTobler, 2017-06-16:
It should be possible to set the parameter `SENS_EN_LL40LS` to 2 (I2C) and not just enabled and disabled.

so it should be:
* 0 -> disabled
* 1 -> PWM
* 2 -> I2C

mhkabir, 2017-06-16:
Looks like loading the parameter meta-data is broken. QGC doesn't load the new metadata anymore when flashing a board (either the stable build from online, or using a custom firmware file)

DonLakeFlyer, 2017-06-16:
I would guess the parameter version was not bumped up on the firmware side. Hence QGC thinks it's the same version and will not overwrite.

DonLakeFlyer, 2017-06-16:
Can someone on the firmware side check that. Re-open if the parameter version was increased and it is still not working

closed 2017-06-16 by DonLakeFlyer