mavlink/qgroundcontrol #5305: Log download incorrect file extension

dagar, 2017-06-17:
Using PX4 and QGC master logs are downloaded with the .px4log extension, but these are from the new logger (.ulg).

dagar, 2017-06-26:
This seems to be confusing a lot of people.

DonLakeFlyer, 2017-06-27:
@dagar Is there some way to know what type of log file they are? Based on firmware version or something like that? How to handle old/new firmware?

dagar, 2017-06-27:
I take it that means you can't get the file extension through mavlink?

The logger is set based on the SYS_LOGGER parameter. The new format is default. https://dev.px4.io/en/advanced/parameter_reference.html

[image]

DonLakeFlyer, 2017-06-27:
Ok, looking at the current code I"m confused:
```
        // This is a stopgap and should be removed once log file types are properly supported by the log download protocol
        if (_vehicle->parameterManager()->getParameter(FactSystem::defaultComponentId, "SYS_LOGGER")->rawValue().toInt() == 0) {
            _downloadData->filename += ".px4log";
        } else {
            _downloadData->filename += ".ulg";
        }
```

DonLakeFlyer, 2017-06-27:
I also just tried to download a log using master SITL and the SYS_LOGGER parameter was not there.

dagar, 2017-06-27:
It's a parameter that's only used in an init script to decide which logger to start, so I guess it's not being set active.
Can you default to ulg unless SYS_LOGGER is available and set to 0? I imagine in a few releases the old logger (and this param) could disappear completely.

DonLakeFlyer, 2017-06-27:
Sure. But I don't get why people are seeing problems? Are they reporting this from a stable build. Not sure when the change to support .ulg went in.

DonLakeFlyer, 2017-06-27:
Also the strange thing with SITL as that there were logs there that I could download. Even though SYS_LOGGER was not there. Does that make sense?

closed 2017-06-27 by DonLakeFlyer