mavlink/qgroundcontrol #5311: Crash when navigating Vehicle Setup tabs under Windows

jaxxzer, 2017-06-20:
I've heard two similar reports about an application crash when switching between tabs in the Vehicle Setup page on Windows with recent builds. I've experienced this twice myself, but have not been able to reproduce it.

Not sure what's going on, or if it is even a problem with QGC, but I'm putting this here to see if a trend emerges.

jaxxzer, 2017-06-20:
Here is an error that is reported to appear:

(QGroundControl.exe:5768): GLib-ERROR **:gmem.c:103: failed to allocate 3110491 bytes

Microsoft Visual C++ Runtime Library
Runtime Error!

Program: C:\Program Files
(x86)\qgroundcontrol\QGroundControl.exe

NaterGator, 2017-06-20:
Glib? That sounds suspiciously like gstreamer...

4tune500, 2017-06-20:
Hi,
Update firmware on the Pixhawk and this should fix it.

jaxxzer, 2017-06-20:
@4tune500 I'm pretty sure you are dealing with a separate issue.

What firmware were you on, and which firmware did you move to?
What's the QGC version you are using.

4tune500, 2017-06-20:
I downloaded all the latest versions (why and firmware) and hey presto it started working. I can't remember the specific version but it was fairly old.

DonLakeFlyer, 2017-06-28:
Firmware could cause the problem: It could be caused by a NULL reference to a missing parameter fact. But if that were the case it would be 100% reproducible for anyone hitting this.

DonLakeFlyer, 2017-07-07:
Re-open is it comes back

closed 2017-07-07 by DonLakeFlyer

HanJIk, 2017-09-15:
I am experiencing the same phenomenon.
Resolution??

jaxxzer, 2017-09-15:
@HanJIk 
What firmware?
What version QGC?
Which buttons are you clicking?

DonLakeFlyer, 2017-09-15:
I believe I fixed a setup page crash about a month ago. Newer builds should have it, including latest Stable. Need to know versions you are using.

HanJIk, 2017-09-17:
firmware ardusub3.5.1
QGC version Download from ardusub homepage to download on 20170915

DonLakeFlyer, 2017-09-17:
@jaxxzer Windows builds can capture stack traces and be debugged for crashes. @NaterGator Should be able to help you set it up. As far as setup, would be good to get the steps into the QGC devs docs one  you get the info.

jaxxzer, 2017-09-20:
Hey @NaterGator I'm in touch with someone who is seeing this fairly regularly and has offered to help test.