mavlink/qgroundcontrol #5313: How Can I Video Streaming using MAVLink?

HTaeHyeon, 2017-06-20:
Hello, There.
I have some problems about Video Streaming.
I saw this page, and I success Video Streaming using gstreamer.
(https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md)
Now, I have to stream using MAVLink. So I read the mavlink website and I found that.
[image]
but In the source, there aren't someting about VIDEO_STREAM such as SET_VIDEO_STREAM_SETTINGS, MAV_CMD_VIDEO_START_STREAMING etc...
How Can I Video Streaming using MAVLink?? Please tell me.

lbegani, 2017-06-21:
I do not think QGC supports video streaming over MAVLink. However there is a work in progress where video streaming details will get shared with GCS over MAVLink messages. Once the QGC has the URI from the vehicle, it will start video streaming with the specified URI over UDP/RTSP protocol
https://github.com/mavlink/qgroundcontrol/pull/5271
https://github.com/mavlink/mavlink-devguide/pull/9

DonLakeFlyer, 2017-06-21:
Issues are not the place to ask questions. Use the QGC forum for that: http://discuss.px4.io/c/qgroundcontrol

closed 2017-06-21 by DonLakeFlyer