mavlink/qgroundcontrol #5316: Poor user experience when changing video source/setting

SaliniVenate, 2017-06-21:
Currently, the video source can be selected in the “Settings View” and the video can be seen in the “Fly View”. To change the video source/settings while viewing the video, we need to switch from the “Fly View” to “Settings View”. This results in a bad user experience. We should be able to change the video source/settings and watch the video at the same time.

It would be nice if we can have a way to select the video source and related settings from the “Fly View” itself. Need comments and suggestions. 

[image]

[image]

SaliniVenate, 2017-06-21:
@DonLakeFlyer , @dogmaphobic , @julianoes , @lfelipe, @lbegani,

SaliniVenate, 2017-06-21:
@dogmaphobic , @DonLakeFlyer ,  @julianoes , @lfelipe , @lbegani

DonLakeFlyer, 2017-06-21:
I don't see this as being an important change. The same could be said for every Application Setting. They all relate to same part of the user interface.

lfelipe, 2017-06-21:
@DonLakeFlyer we talked a bit about this during the PX4 dev call earlier today with @dogmaphobic as well, it looks like he was already interested in doing something like this as part of his work on the video section. 

Currently on #5271 there is a basic implementation of a separate widget to do this to allow users to modify video settings while actually seeing the video, but it was mentioned that this particular implementation would not directly work on mobile platforms (which is the reason why @SaliniVenate opened this issue here, to ask for more feedback on that).

closed 2026-10-04 by github-actions[bot]