mavlink/qgroundcontrol #5321: Radio Setup > Additional Radio Setup > Copy Trims text is confusing

hamishwillee, 2017-06-22:
When doing "copy trims" on PX4 it says 

Center your sticks and move throttle all the way down. Then press OK to copy trims. After pressing Ok, reset the trims on your radio back to zero.".

The last sentence doesn't make sense because the instructions don't suggest changing the trims on the radio in the first place. 

The copy-trims setting does not appear for ArduPilot.

DonLakeFlyer, 2017-06-27:
This text is actually correct. The point of Copy Trims is to move the manual radio trims you have set on your transmitter to the autopilot, then they are no loonger needed on the transmitter.

hamishwillee, 2017-06-27:
OK. Reopening as a placeholder to fix the *documentation* ... since how this actually works still isn't clear to me.

What we say now in the docs is 

* "Copy Trims is used to save the zero point values for your sticks to the appropriate RC parameters (i.e. you set the trims to zero and then copy the current values to the associated parameters)." 

What I think we need to say is (?):

* Copy Trims is used to copy the manual trim settings from your transmitter so that they can be applied automatically within the autopilot. After this is done you will need to remove the manually set trims.

Does that make sense.

In addition, can you explain how this actually "works"/what is copied? It sounds like the trims must be separately sent to the receiver by the transmitter and mixed in the receiver - otherwise how can you infer copy them? .... since you're coping/calibrating when the sticks already have the offset applied.

DonLakeFlyer, 2017-06-28:
> Copy Trims is used to copy the manual trim settings from your transmitter so that they can be applied automatically within the autopilot. After this is done you will need to remove the manually set trims.

That is correct. 

As far as how it works internally to the firmware you'll have to ask firmware folks for exact details. I does some futzing with parameters but can't remember excatly how.

hamishwillee, 2017-06-28:
Thanks. I fixed this up and merged (https://github.com/mavlink/qgc-user-guide/pull/91). No need to explain exactly how it works, just what it does.

closed 2017-06-28 by hamishwillee