mavlink/qgroundcontrol #5323: QGC:Not enough contrast between text and background on airframe selection

pug398, 2017-06-22:
Pic to hopefully show what I am talking about. Difficult to see white text on light grey background and impossible to see on yellow highlight. At least on my big monitor.
[image]

DonLakeFlyer, 2017-06-26:
Can you try a newer daily build. This is working fine for me.

DonLakeFlyer, 2017-06-26:
It's fine to trun daily build of QGC with stable builds of firmware.

pug398, 2017-06-26:
Sorry I am just a little skittish now silly I know. I think previous problem resulted from QGC>Settings>Parameters>"reset all parameters" with ardupilot and it said it failed. I then set what I wanted manually and tried to fly. Probably created some bad artifacts and didn't get it straightened out till I put PX4 back on it and reset all parameters. I will check out QGC daily build and thanks for looking into it.

pug398, 2017-06-27:
@DonLakeFlyer same issue with daily build. Can't read white text through yellow background but it is no biggy just have to remember to read it before highlighting it. I note that vehicle setup buttons on left change text to black color when yellow highlighted an text is easy to read.

DonLakeFlyer, 2017-06-27:
Thanks for retesting. I just realized I tested it using the wrong build. I can reproduce this now myself.

closed 2017-06-27 by DonLakeFlyer