mavlink/qgroundcontrol #5337: VTOL: Transition action or menu item

sanderux, 2017-06-24:
@DonLakeFlyer managing a VTOL solely through QGC is missing one important feature;
The user must be able to set the flight mode (Hover or Forward flight)
Ideally it is shown next to Arm status by showing the current mode and clicking it to change mode.

Getting state: https://pixhawk.ethz.ch/mavlink/#MAV_VTOL_STATE
Switching state: https://pixhawk.ethz.ch/mavlink/#MAV_CMD_DO_VTOL_TRANSITION

DonLakeFlyer, 2017-06-25:
I'll take a lot at that in 3.3

sanderux, 2017-09-19:
Hi @DonLakeFlyer 
Do you have any progress on this? It is a highly requested feature for clients

DonLakeFlyer, 2018-01-04:
@sanderux Can you test this once it gets through the build system. I tested with SITL but it seems like the VTOL gazebo support may be strange/buggy? When I transition to forward flight, it sticks for a few seconds then just switches back.

sanderux, 2018-01-04:
i'm not getting any action when switching with the current daily

DonLakeFlyer, 2018-01-04:
Do you get the slider?

sanderux, 2018-01-04:
yes, it seems to be missing sending the actual command in the pr

DonLakeFlyer, 2018-01-04:
Crap there are some changes missing...

PR #5956 (2018-01-04): Support VTOL transition
Fixes: #5337, #5620

[image]
closed 2018-01-05 by DonLakeFlyer