mavlink/qgroundcontrol #5365: Awkward planning vs flight screen

AndKe, 2017-06-29:
I find it very strange to use QGC for missions, 
In order to monitor a mission,  see waypoints and the missions progress, the "planning" window is the only one that can do it. - it does not even contain voltage or mode controls.

The flight view is what I would expect to watch when flying, however, there is no clue as how the mission progresses, no way to jump to another WP, or otherwise to know "where in a mission" the UAV is and when it's expected to do what.

the planning view should have all necessary monitoring and control tools.
or
the flight view should at least display the mission waypoints, and have the mission altitude controls/graph that is on the bottom of the planning view.

To be honest, this is so bizarre, that I kind of believe it could be some bug that only I have seen, but I failed to see everything presented in one place in my attempts.
Seeing a arrow travel across the screen, not knowing when and where the next turn is to be expected, is not fun, nor work, and certainly not having good situation awareness, you see a steep bank angle, and start guessing if that is where a sharp turn was planned..

DonLakeFlyer, 2017-06-29:
Something seems very wrong:
* You should see the mission in the fly view
* The currently active waypoint will change to green as the mission progresses
* You can click on a waypoint to change the currently active waypoint

AndKe, 2017-06-29:
thanks, (good to hear)  - will try to start fresh tomorrow.
today: I planned while solo waited. , did not see it in flight view, uploaded mission, dis/reconnected (by wifi) , flew it whithout seeing any wp in flight view at all. 

Will try again tomorrow, maybe find a way to reproduce.

AndKe, 2017-06-30:
This report was due to https://github.com/mavlink/qgroundcontrol/issues/5370 - which made me believe this was an issue.

closed 2017-06-30 by AndKe