mavlink/qgroundcontrol #5368: Survey: how to resume after battery change

AndKe, 2017-06-30:
If we think of a multirotor survey, since survey poses as only one mission item, how can user land, then resume from correct place in mission ?

Usually we would (for example) fly WP 1...50 , poshold, disable trigging by distance, land, switch battery, take off, fly to wp50 or 51, poshold, enable trigging, continue.

the way survey works, one can't see individual waypoints, or resume at proper place.

AndKe, 2017-06-30:
So far, I am aware of the option to download the mission from vehicle, and when see it as a pile of WP's.

DonLakeFlyer, 2017-06-30:
* Do an RTL from the middle of a mission
* Turn off the vehicle, swap battery
* While doing this do not click Disconnect when QGC says lost communication
* Turn vehicle back on
* QGC iwill reconnect to vehicle
* Resume Mission confirmation will pop up
* Confirm and it will re-generate the mission from the point it left off and start the reset of the mission

DonLakeFlyer, 2017-06-30:
I think that's the same process as Mission Planner

AndKe, 2017-06-30:
Thank you. 
I did some tests today, and discovered that both what you wrote , and the more "old-fashion" method of 
 skipping to a WP works fine,  (I were afraid missions were collected and visible as survey only, not as individual waypoints.)

DonLakeFlyer, 2017-06-30:
Great, thanks for reporting back that it's working. There is a big difference between using resume mission and skipping to a waypoint. When you resume mission if the following commands:
```
    includedResumeCommands << MAV_CMD_DO_CONTROL_VIDEO
                           << MAV_CMD_DO_SET_ROI
                           << MAV_CMD_DO_DIGICAM_CONFIGURE
                           << MAV_CMD_DO_DIGICAM_CONTROL
                           << MAV_CMD_DO_MOUNT_CONFIGURE
                           << MAV_CMD_DO_MOUNT_CONTROL
                           << MAV_CMD_DO_SET_CAM_TRIGG_DIST
                           << MAV_CMD_DO_FENCE_ENABLE
                           << MAV_CMD_IMAGE_START_CAPTURE
                           << MAV_CMD_IMAGE_STOP_CAPTURE
                           << MAV_CMD_VIDEO_START_CAPTURE
                           << MAV_CMD_VIDEO_STOP_CAPTURE
                           << MAV_CMD_DO_CHANGE_SPEED;
```

are prepended to the remainder of the mission so that you initial setup is kept intact. It's a little more complicated in reality (dup removal and so forth). But the simple example is if you changed flight speed prior to where you are resuming the mission, the change flight speed will prepended so the mission remainder is flown at the right speed.

AndKe, 2017-06-30:
Thanks, I discovered that it had more logic to it, I did not spent too much time on that, but to me it seems like a paused mission disabled MAV_CMD_DO_SET_CAM_TRIGG_DIST  - which is nice.

What I also discovered, is that flying between  WP30 and 31 in a survey - if mission were paused (the UAV moved away) , then resumed it would not restart with 30, then go to 31 , but rather fly directly to 31 (missing a bit of the survey area.)

Usually, I had to position the UAV and stop/resume trigging to have a clean dataset between battery changes. This is great.
Is there a way to make it go to WP-1 when resuming ?

DonLakeFlyer, 2017-07-01:
Hmm, yeah teh point at which it resumes seems wrong. I'll back it up another waypoint.

DonLakeFlyer, 2017-07-07:
> What I also discovered, is that flying between WP30 and 31 in a survey - if mission were paused (the UAV moved away) , then resumed it would not restart with 30, then go to 31 , but rather fly directly to 31 (missing a bit of the survey area.)

I was wrong, the code already backs up a waypoint. I ran it through both PX4 and ArduPIlot firmwares and with both the resume waypoint was correctly the one before the item it was moving towards. I am specifically talking about Resume Mission which will only pop up if you land from the middle of mission. 

There is no support for Pausing a mission, and then without landing, having it automatically go back a waypoint when you "Continue Mission". Which now that I look at what you wrote again seems like the scenario you are talking about.

Assuming you were not talking about Resume Mission (which is a battery change scenario) then this is working correctly. I'm not sure having Continue Mission while in flight makes sense to always autoatmcailly go back a waypoint. That really depends on what you did while you were out of mission mode which there is no way to know about. So I'm going to leave that as is.

closed 2017-07-07 by DonLakeFlyer