mavlink/qgroundcontrol #5370: Mission upload stuck

AndKe, 2017-06-30:
Uploading a mission to 3DR Solo, works well, it ends with the yellow "flight plan received" message.
the sync button, however, continue to rotate. there is no way to reset, download/upload or load/save until QGC is restarted.

AndKe, 2017-06-30:
Re-tested that a lot today. (using Solo only, as it and RX100 camera was the testing object) 
It's perfectly reproducible.

- Start QGC (master)  (in this case Ubuntu desktop)
-load, or create a survey
-connect 3DR solo (latest original Solo firmware, not based on fresh ArduCopter)
-in upper right corner, there will be a button that urges to upload, use that *OR* the "Sync/upload" button
-observe a huge DONE / and "flight plan received" 
-the sync button spinning animation continues, the button is now useless.
-the "flight view" does not display the uploaded mission.

-close/open application
-flight view displays correctly the complete mission that was uploaded.
-same problem will spawn again if anything is attempted uploaded to solo.

DonLakeFlyer, 2017-06-30:
Thanks, hopefully I can repro on my Solo.

AndKe, 2017-06-30:
if not, I'll send you tlogs, there is most likely a difference in what the AP say at the end of transmission.

DonLakeFlyer, 2017-06-30:
It repros for me

DonLakeFlyer, 2017-06-30:
GeoFence bug. Also the likely reason why the Fly view isn't showing mission correctly some times.

AndKe, 2017-07-04:
I can still reproduce it at master b682031e7  
to reproduce: 
-connect to a Solo with a mission on board
-in planning view, download mission
-upload mission
-the spinning animation will not stop.
[image]

Otusbubo, 2017-10-18:
Did some in-office testing yesterday, and experienced this problem on both an iPad and an Android tablet.  Worked around it by uploading, experiencing the "sticking," closing qGC, reopening, then downloading the mission from the Solo.  Will flight test later today.

DonLakeFlyer, 2018-01-04:
I tried this on latest daily build and it is working fine for me now. Can you folks retest with latest daily and I'll reopen if still there.

closed 2018-01-04 by DonLakeFlyer