mavlink/qgroundcontrol #5371: Mission config suggestion.

AndKe, 2017-06-30:
Mission has a "flight speed" option.
please consider adding a "survey speed" option to survey.  whenever cam_trig_dist > 0 , apply "survey speed"  

Travelling to and from survey and between surveys at different altitude is best done at high speed, while photographing is restricted by exposure time to prevent motion blur.

DonLakeFlyer, 2017-06-30:
Yeah. This has been requested from multiple people. Will get into 3.3. For now:
* Set the faster flight speed in Mission Start
* Add an additional waypoint just prior to entering the survey area and slow down the flight speed on that waypoint. (Flight speed changes are now available on waypoint items)
* Add an additional waypoint past the survey exit and speed back up

AndKe, 2017-06-30:
Thanks, yes, I know - have been planning manually for years using mavproxy / APM planner - so I am very familiar with  the DIY solution, and glad to hear it's planned.

closed 2017-06-30 by AndKe