mavlink/qgroundcontrol #5376: QGC showing build errors

lakshaydang, 2017-07-01:
Hey friends,
I was trying to build the QGC using qt creator but its showing me error.
Previously 4 month back i was building it using qt 5.5.1 but now it is written that the updated code will work in Qt 5.7.1 so i downloaded it. 
Then when i am building it but it is showing me 2 errors
1. Cannot read C:/Users/computer/Downloads/qgroundcontrol/qgroundcontrol/user_config.pri: No such file or directory
 2. Error while building/deploying project qgroundcontrol (kit: Unnamed) When executing step "Make"

I have tried various permutation and combinations but still its not working.
Can you please help me with this?

Thank You

dogmaphobic, 2017-07-01:
Build questions, or non bug related questions in general should be posted on [Guitter](https://gitter.im/mavlink/qgroundcontrol)

>Cannot read C:/Users/computer/Downloads/qgroundcontrol/qgroundcontrol/user_config.pri: No such file or directory

That's an optional file you may create with your own definitions to change the build. Not having it is normal.

>Error while building/deploying project qgroundcontrol (kit: Unnamed) When executing step "Make"

`(kit: Unnamed)` means your Qt installation was not done properly.

closed 2017-07-01 by dogmaphobic

vishal-ahirwar, 2022-06-14:
I'm getting this issue?????
**Project MESSAGE: Qt version 6.2.4
[Inexact] Cannot read /media/vi/510GB/qgroundcontrol/user_config.pri: No such file or directory**

junwoo091400, 2022-11-15:
> I'm getting this issue????? **Project MESSAGE: Qt version 6.2.4 [Inexact] Cannot read /media/vi/510GB/qgroundcontrol/user_config.pri: No such file or directory**

As @dogmaphobic stated above. That is not an issue, and it's normal.