mavlink/qgroundcontrol #5379: Mac OS Daily build broken (Qt Location lib missing)

LorenzMeier, 2017-07-02:
This is on 10.12 (fresh install):
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
```
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtLocation.framework/Versions/5/QtLocation
  Referenced from: /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
  Reason: image not found
```

dogmaphobic, 2017-07-02:
That’s the installer build phase. This has been going on and off for some time. You run 10 builds in a row and they are all fine. Then eventually one messes up the relocation phase (the installer build) for no reason. The error above:

>Library not loaded: @rpath/QtLocation.framework/Versions/5/QtLocation

Shows that the QtLocation library was not properly “relocated” so the app can’t find it.

There are plenty of issues (and fix attempts) about this already.

DonLakeFlyer, 2017-07-02:
Crap

DonLakeFlyer, 2017-07-07:
I'm assuming Gus' change fixed this

closed 2017-07-07 by DonLakeFlyer