mavlink/qgroundcontrol #5381: Pixhawk PX4 Is Dead!

stevec5000, 2017-07-02:
3D Robotics has gone out of business and quit making and selling the PX4 last year after the company crashed due to the failure of their drones!  They don't have accessories or cables or anything for them either and of course they never had support.  When I contacted them to ask if they could fix mine they said: "NO, BUY A NEW ONE!".  I didn't understand it at first but it makes sense that they wouldn't want to take in any for repair since that would open them up to Fraud lawsuits when they wouldn't be able to show they fixed it since it was never able to fly!
So are you guys going to continue to waste your time trying to get this crappy program to run or are you going to give up in defeat after having wasted 3 years of everyone's time?

https://www.forbes.com/sites/ryanmac/2016/10/05/3d-robotics-solo-crash-chris-anderson/#881abff3ff5b﻿

DonLakeFlyer, 2017-07-02:
Go back to your mom's basement steve.

closed 2017-07-02 by DonLakeFlyer

stevec5000, 2017-07-07:
That's not nice!  That's why nobody likes you!