mavlink/qgroundcontrol #5383: No connection using Sik radio via Android and USB/OTG

rsilk1949, 2017-07-02:
I am a new user who is looking for a good GCS for Android/Sik radio setup.
First off this configuration works with Tower.

The USB/serial device is not automatically recognised so I tried adding a link. I did not see an choice for USB or OTG, so I asumed Serial. Baud rate was set to 57600. When I hit the Connect button, I get an error "Error on link. Error connecting. Could not create port. No such file or directory"

Also, I could not find a connect button except in the Settings/Comm Links page. I did see a screenshot of the Mac and app which did have a connect button on the main page. This should be available in QGC IMO.

DonLakeFlyer, 2017-07-02:
Are you using a daily build? If not, then try that. If that still doesn't work we can discuss further from there.

https://docs.qgroundcontrol.com/en/releases/daily_builds.html

rsilk1949, 2017-07-03:
Like many potential users, I have no way to install Android apps except via Google  Play. So I will have to wait for the  next release version.

DonLakeFlyer, 2017-07-03:
The daily build is available via Google Play. Follow the link I posted above.

rsilk1949, 2017-07-08:
Installed beta version and still get the same result.

kikislater, 2018-04-25:
Hi, 
I have the same behaviour ! Video to understand below : 
https://youtu.be/RktFWS_NyjE

DonLakeFlyer, 2018-04-29:
When the popup came on the screen when you plugged in the radio you clicked Cancel. Which in turn did not give acess for QGC to talk with the radio. When that pop up comes up you need to click the QGC icon in the popup.

kikislater, 2018-04-30:
No, it's wrong. Even with cancel, Tower have access to usb serial whereas qgc not !
I could remake a video. It will not give the other popup to grant authorisations. Difference between Qgroundcontrol and Tower is that Qgroundcontrol is not allowed to modify system parameters.

[image]

[image]

First paramter in Tower shows is 'modify system parameters'. I have other app like drone deploy, dji go4, drone harmony and I have this parameter in app and popup to grant authorisations

DonLakeFlyer, 2018-04-30:
I still don't understand if you click on QGC from the Android popup?

So you plug in the SiK radio. The Android popup comes up asking you what app to use with the radio: Tower or QGC. You click QGC, QGC should launch. If there is a vehicle on the other end of the radio connection (which it seems like there was due to red flashing on radio from video) then QGC will automatically bring up that vehicle. You don't need to do anything with comm links.

DonLakeFlyer, 2018-04-30:
If the above doesn't work turn on LinkManagerLog and MultiVehicleManagerLog: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html. Save the console log output to a file and attach here.

kikislater, 2018-04-30:
Ok so let's watch this video. Nothing in logs except when I force com link 
https://youtu.be/xxHeMMNLIBo

DonLakeFlyer, 2018-04-30:
Two things:
* You didn't turn those logs on in the console
* I'm not sure what the interaction is with Tower and USB since you had the Tower app paused on the device when you launched QGC. There may be some bugs associated with that combination. It may still have had some sort of hold on the USB connection. Probably not, but best to remove that possibility from testing.

Can you try again, making sure you close any paused Tower app from the device, but also turn on LinkManagerVerboseLog.

DonLakeFlyer, 2018-04-30:
Also what version of QGC is this?

kikislater, 2018-04-30:
I will do ! But I ever test and it's not working. If Tower were still active, you will see 2 tower icons in notification area.
It's stable version from playstore.

DonLakeFlyer, 2018-04-30:
I've been flying with my Samsung Tablet and a SiK radio almost every day for the last couple weeks. Hence my thinking it has something to do with Tower sitting around inactive (as can be seen at 1:24 in the video). Since that is the only difference that I can see from what I am doing.

kikislater, 2018-04-30:
Where the logs are stored in Android version ? 

New video here : https://youtu.be/KGv4ezaMZ9c

But may be it works with your device only and not with others. Other user on this thread have problem. If I remember well, previous build was working !

DonLakeFlyer, 2018-04-30:
@AndKe You're using andoird aren't you. Are you using a SiK Radio?

kikislater, 2018-04-30:
Should be related to my device in that case. I will experiment my wife has the same android model ... 
It's boring but I want to understand what is the problem !

DonLakeFlyer, 2018-04-30:
Thanks @AndKe 

@kikislater I still think it's related to the Tower install somehow. If you have another device which does not have Tower on it would be good to try on that. Also if Stable doesn't work. Can you try Daily  as well?

kikislater, 2018-04-30:
Yes I was on daily 3.2 before if I remember ... 
Interesting thing is that I have nothing in qgroundcontrol/telemetry. Folder qgroundcontrol and subfolders are here but nothing inside.

kikislater, 2018-04-30:
Tower removed, qgc daily installed ... It works !

kikislater, 2018-04-30:
Réinstallation of tower brick qgc usb serial connection :-/

DonLakeFlyer, 2018-04-30:
Helpful, thanks.

rsilk1949, 2018-05-01:
After reading the above activity, I deleted Tower *and* the old QGC. I removed all supporting data files I could find using File Manager.
I then installed QGC 3.3.2 from the Google store.
I have not yet attempted to connect to the radio on the Pixhawk.

When I plug in the OTG cable, I do not get asked by QGC to use the USB port. Using the "Easy OTG" app, I can see the FTDI unit (radio). I also checked my radio serial baud rate is 57600 using the 3DR exe.
When I check the Comm Links screen, there is no port displayed, so no chance to try a connect.

I have a report on the USB device (FTDI) from my Mac:
FT230X Basic UART:

  Product ID:	0x6015
  Vendor ID:	0x0403  (Future Technology Devices International Limited)
  Version:	10.00
  Serial Number:	DA00BHVJ
  Speed:	Up to 12 Mb/sec
  Manufacturer:	FTDI
  Location ID:	0xfa130000 / 5
  Current Available (mA):	1000
  Current Required (mA):	90
  Extra Operating Current (mA):	0

AndKe, 2018-05-02:
"I believe current manifest should prompt on any USB device at all plugged in. "
I can confirm that it does, annoyingly as always.
Whatever I plug in an thermal camera, logic analyzer, endoscope or GearVR, - QGC pops up.  - I would rather prefer it only offered to grab a device when running only - if possible.

rsilk1949, 2018-05-02:
It looks like I did not correctly uninstall Tower, sorry - my error :(
After correctly uninstalling it, I also uninstalled QGC. I then installed QGC from the Play store for a second time.
Now when I first launched it and inserted the OTG cable, I did get prompted to allow QGC access to the USB device :)
Remember, in this scenario I still do not have the radio on the quad switched on.

I can now see this error in the console:
[E] at qrc:/qml/SerialSettings/.qml:75 - "Serial Port not present"
When I try to manually connect, an error message pos up "Error on link: Error connecting: Could not create port. Socket operation on non-socket"

Interstingly, the red light on the radio is blinking showing some sort of activity.

Are these issues due to not having the receiving radio on?

rsilk1949, 2018-05-02:
My apologies, but it looks like I did not correctly uninstall Tower.
Now when I first launched QGC 3.3.2, it asked me to attach the OTG to the app :)
In the console log, I see
[E] at qrc:/qml/SerialSettings.qml:75 - "Serial Port not present"
If I try to manually connect, I see error msg
Error on link. Error connecting: Could not create port. Socket operation on non-socket.

Intrestingly, I can see the red light flashing on the radio indicating some communication activity. Remember, I do not have the second radio powered on.

kikislater, 2018-05-02:
You try to connect to another device which is not connected ? 
Don't understand your input ! You should ry in real situation IMO ...

rsilk1949, 2018-05-02:
Success!
When I also power on the pixhawk and radio on the quad, I get an auto-connection.
HUD, vehicle summary, etc. works as expected :)
I even deleted the serial device in the Comm Links page and everything continues to work with no configuration required. Impressive.
Looks like QGC, with it's active support and developmemt, will now be a good solution for my in the field needs . . . .

kikislater, 2018-05-02:
Good to know ! 
So in both cases tower is the problem. I solve problem also with others devices for others hardware. Interesting to know that an app could block another but also sad :-/  Android is not as good as I thought !

DonLakeFlyer, 2018-05-02:
> Interesting to know that an app could block another but also sad

That usually isn't the case. The problem tends to be at some point you click the "Always use this app with this connection" without remembering. And then you are hosed with that app sucking up that usb connection for all time.

I'll do some testing to verify that is the case.

AndKe, 2018-05-02:
> Always use this app with this connection" without remembering
Yes, that's a easy mistake, especially for Android builds a bit back that offered "always" as default action.
It's easy to verify in "settings"->"app" and both apps.  However, @kikislater did uninstall/reinstall Tower, and this action resets any thing like that.

I still suspect it to have something to do with the nasty way QGC attaches to everything.  I have plenty of phone connected devices, and always get only the relevant options - and it works. 
Except for QGC - that will attach to everything it sees.

Once I had a defective vibration-logger/analyzer app (some sort of serial adapter) that suggested several connection right on the boot of the device (with nothing attached)  another Android release (think it was 6.0.1 or thereabout ) somehow removed the nagging about connection on boot, and the app stopped seeing the connected device altogether (even after restoring known good .apk) 

*Maybe* there is some logic in some Android builds that lower priority of apps that try to connect wildly to anything, favoring software that is more specific.

> Android is not as good as I thought !
...or there is something very different with QGC. - It could at least have options to select device-types it will try to connect to. - or offer connection only when it's running.

p51mustang, 2018-05-07:
Help.  My SIK radio works marvelously when I use it to connect my Mac QGC and my plane.  When I've tried the same on a brand new Samsung android tablet I have no luck.

Specifics are:

1.  On the Samsung and Mac, the green light goes solid on the radio.  Red light starts to flash.  
2.  On the Mac, both orange lights flicker, indicating send/receive
3.  On the Samsung, only one orange light flickers.  

No connection to remote radio / plane on the Samsung, but consistently on the SIK radio.

Looking at Mac QGC, here's what I see for the com to this device:
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:495 - "portName:           "cu.usbserial-DM004MH6""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:496 - "systemLocation:     "/dev/cu.usbserial-DM004MH6""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:497 - "description:        "FT230X Basic UART""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:498 - "manufacturer:       "FTDI""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:499 - "serialNumber:       "DM004MH6""
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:500 - "vendorIdentifier:   1027"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/LinkManager.cc:501 - "productIdentifier:  24597"

On the Samsung, I see that the connect called is successful (SerialLinkCC:121), that the link succeeds (SerialLinkCC:167) and that the connection is set (SerialLinkCC:253).  That's it though - there is no double orange flash.

I've used the SIK radio config tool to verify both radios are communicating.  

Please advise

DonLakeFlyer, 2018-05-12:
I've been doing testing in combination with Tower. I looks like the way Android works is that if you give a connection to an app it will retain that connection until you close that app. That makes sense. So if Tower has the radio connection it keeps it until you completely close Tower. The problem is that I've found that if Tower is dormant but open on the device and then you plug a radio in Tower tends to crash when it tries to re-open. The screen will flash and then nothing will happen. This then leads you to think that Tower does not have the connection associated with it. But it still does and you can't use QGC with the radio until you then close the Tower app.

kikislater, 2018-05-13:
It's not the same for me. If I installed tower, reboot device. And try to make qgc working, it doesn't work ! 
But I suspect in my case that Xiaomi update break some things with otg ...

DonLakeFlyer, 2018-05-14:
And uninstall Tower makes it work?

kikislater, 2018-05-15:
Yes that's it ! 
But saturday I had an update from xiaomi and at this time OTG is break ...

DonLakeFlyer, 2018-05-23:
It seems like newer version of android remembers to app you associated a device with. So you you plug in a radio and then pick Tower then Tower will hold on to that in settings. QGC ends up doing the same thing. This seems really bad. Not sure how to work around that at the moment. Closing the app doesn't reset it. You need to go in and clear defaults in Settings, App, Set a Default.

kikislater, 2018-05-24:
Killing app work for me but I need sometimes to reboot phone. But I found also that I have problem with USB. Time to consider hc05 / hc06 bluetooth modules !

DonLakeFlyer, 2019-01-21:
https://developer.android.com/guide/topics/connectivity/usb/host

I think QGC needs to request permission itself to use the USB if it does not already have permission because some other app has the usb set to it.

DonLakeFlyer, 2019-01-23:
I'm hoping I managed to fix this yesterday. I tested with QGC and Mission Maker installed. Can someone on this issue give it a try to verify?

kikislater, 2019-01-30:
Well, I will test next week ! 
I will keep you updated. Thanks

robeod7, 2019-02-20:
I cannot get QGC to connect to pixhawk using 3dr sik radios. I do not have Tower or any other app installed. The radios work fine using mission planner on PC.
I have a solid green lights and blinking red lights on both radios.
I've tried the regular app and the daily test build on Android 8.0.0 on a Samsung Galaxy S7.
I attached my log, please help!
[Log.Txt](https://github.com/mavlink/qgroundcontrol/files/2885982/Log.Txt)

DonLakeFlyer, 2019-02-20:
Are you prompted by Android to allow QGC to use the device?

robeod7, 2019-02-20:
Initially yes. I remember years ago when i was using the Tower app with the same radio's and an older Android phone, i had to open a separate app called 3DR services or something to get Tower to work. I've tried QGC on my tablet as well with the same results. Any idea what i'm missing?

p51mustang, 2019-02-21:
Yes - Android does prompt to allow QGC to use the device (once or always are the choices)

DonLakeFlyer, 2019-02-21:
This link https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html will tell you what logging to turn on for debugging connection problems. Follow that and attach the log.

robeod7, 2019-02-21:
here is my console log
[Newlog.txt](https://github.com/mavlink/qgroundcontrol/files/2891676/Newlog.txt)

DonLakeFlyer, 2019-02-22:
@robeod7 As far as I can tell from the log output QGC is connected to your SiK radio. It just isn't seeing any traffic on it which inidicates there is a vehicle on the other end of the link.

DonLakeFlyer, 2019-03-22:
I"m pretty sure this is all fixed now

closed 2019-03-22 by DonLakeFlyer