mavlink/qgroundcontrol #5388: "Start Mission" doesn't always get PX4 fixed wing into the correct launch detection states

Antiheavy, 2017-07-03:
Aircraft will sometimes enter Mission mode and Armed state, but Launch Detection will not be running. Picking up and moving the aircraft may trigger an exit from Land Detector and the motor will spool up. This is dangerous.

This seems to happen when something goes wrong with the QGC "Start Mission" slider commands and the vehicle seems to reject something initially, but then enters Armed state anyway.

FYI - @dagar @DonLakeFlyer

Antiheavy, 2017-07-03:
here are some notes we took the most recent time this happened:

Start Mission would not arm the vehicle (error message after many seconds "vehicle could not be armed" or something like that.
shut down QGC
unplugged SiK radio
plug in SiK radio
reload QGC
reconnect to aircraft, it is in Return mode
reboot aircraft
boots up in Hold mode unarmed
lots of preflight errors as mission is syncing and gps is aquiring
after a time things look okay
then Start Mission
switches to armed and Mission, but no launch detection
pick up aircraft and motor spools up dangerously!
disarm and rearm then launch detect running.

DonLakeFlyer, 2017-07-03:
From @dagar:
There's some mission feasibility weirdness on the PX4 side, but perhaps more importantly we need to look at the start mission slider in QGC.

One option is to have it use MAV_CMD_MISSION_START which only arms if you have a valid mission, and can switch into it successfully. Importantly it's a single command and you can specify the starting waypoint.

The other option is to look at what it's doing now and make it a little more robust. Switch to mission mode first, verify, then arm.

DonLakeFlyer, 2017-07-03:
@dagar Would it be possible to change the spec/firmware for MISSION_START such that last_item can be -1 which means run to end? That would make it easier to use. I checked ArduPilot and they don't support first or last item so it doesn't affect them.

DonLakeFlyer, 2017-07-06:
@LorenzMeier Should I change to use MISSION_START command to do a Start Mission from guided action ui?

DonLakeFlyer, 2017-07-07:
I tried reversing the order of change to Mission flight mode and then Arm. That didn't work. It let me switch to Mission flight mode even though the mission was bad (first waypoint too far). Since the flight mode changed, it then happily armed the vehicle as well.

@LorenzMeier @dagar I can't use the MISSION_START command either since it requires specifying the start and stop mission item. I don't have that information readily available where this is needed. Also Start Mission guided is used to also start missions which have had their starting mission item changed by the user. I need a version of MISSION_START which can take -1 for start/stop to signal use the currently set item and mission length. Then I could use if for the Start Mission guided ui. All of this (in combination with the firmware issue above) leaves me confused if there is something I should change from the QGC side.

dagar, 2017-07-07:
Weird, I swore I left a comment here earlier.
In PX4 I only implemented first_item for MAV_CMD_MISSION_START and even that argument I didn't really need. Making them both optional or even dropping them would be fine.

I'm not quite following the QGC mission guided ui problem. What's the situation where you're starting midmission?

DonLakeFlyer, 2017-07-07:
> or even dropping them would be fine.

Dropping then would match current ArduPilot state of things.

> What's the situation where you're starting midmission?

Use can change the current item in a mission by clicking on it. Then you do a "Start Mission".

DonLakeFlyer, 2017-07-07:
The additional case is "Continue Mission" where the user was flying a mission and then Paused or manually took it out of Mission flight mode. Would be good if I could use MISSION_START for that as well is possible. If not for that case I can still just flip to Mission flight mode.

dagar, 2017-07-07:
Isn't the existing MISSION_START perfect for this? It also checks the index before changing to mission (if valid) and arming. Sending MISSION_SET_CURRENT first for Ardupilot would be harmless for PX4.

One thing to note though is we might actually want to be careful about how you're allowed to start a mission. There are many cases where not starting from the beginning is invalid. FW skipping proper takeoff or VTOL skipping transition.

What are we missing in Mavlink or PX4 for QGC having a better understanding of mission validity?

DonLakeFlyer, 2017-07-07:
The current mission item when you do a continue mission is whatever the firmware things it is. It doesn't reset it and the continue the mission. The point is a don't want to have to pass around mission item indices all over the place on the QGC side to use MISSION_START since the firmware already had the correct information.

> What are we missing in Mavlink or PX4 for QGC having a better understanding of mission validity?

Everything. QGC knows nothing about mission validity. There is no anything which supports getting that information back to QGC other than a STATUS_TEXT messages which get blasted across. That said I don't need a solution for that right now I need a solution I can use to get 3.2 out the door in the next couple days.

I would also assume that since the firmware allowed me to change to Mission flight mode with a bad mission on the vehicle. That MISSION_START is going to do the same thing since it seem like the code just pretty much does what QGC does internally (but in reverse order). Changed to flight mode and arm.

dagar, 2017-07-07:
There are a few different cases and the only one where I think you should be passing the index around is when it's selected on the QGC side.

1. You're disarmed on the ground starting the mission for the first time. Send MISSION_START with first waypoint.
2. You're already flying a mission, but want to change the active waypoint. QGC has the index and sends MISSION_START or MISSION_SET_CURRENT
3. Continue mission, I really think we need to be careful here. For a plane or VTOL on the ground arming into mission other than from the beginning can be dangerously wrong. Something like TAKEOFF mode followed by switching to MISSION could work.

MISSION_START first checks that the mission is valid (including start index), then switches to mission mode, and only arms if the transition completed.

DonLakeFlyer, 2017-07-07:
So can I get a version of MISSION_START which accepts -1 for first and last misison item to run?

Antiheavy, 2017-07-07:
BTW - on recent flights we've been seeing this happen roughly half the time when we use the Start Mission slider.   In order to get launch detection to start running we need to disarm and then re-arm.  "Resume Mission" slider pops up when we disarm which is pretty annoying and seems to not work reliably for fixed wing anyway so that may be a related by separate issue.  We need to dismiss the "Resume Mission" slider and either manually Arm or pull up the "Start Mission" slider again to re-attempt it.  On rare occasions it has taken several attempts.

DonLakeFlyer, 2017-07-08:
> "Resume Mission" slider pops up when we disarm which is pretty annoying and seems to not work reliably for fixed wing anyway so that may be a related but separate issue.

Can you put in a separate issue for that. Also, this problem will have no affect on launch detection since QGC knows nothing about launch detection.

DonLakeFlyer, 2017-07-08:
Also if the mission on the vehicle has no problems from a mission feasibility check standpoint then the launch detection problems seem like firmware problems. QGC just arms the vehicle and changes flight mode to mission. Which can easily done by hand as well. Nothing special going on there. I'd post an issue on firmware with the plan and a vehicle log of the failure along with it.

The discussion here is more around what happens when the mission feasibility check fails. And how to make QGC+firmwrae do the right thing in that case.

dagar, 2017-07-08:
PX4 MISSION_START which accepts -1: https://github.com/PX4/Firmware/pull/7557

LorenzMeier, 2017-07-08:
This is the proper fix: It will switch to mission first, then arm - as it should be:
https://github.com/mavlink/qgroundcontrol/pull/5415

DonLakeFlyer, 2017-07-08:
> I tried reversing the order of change to Mission flight mode and then Arm. That didn't work. It let me switch to Mission flight mode even though the mission was bad (first waypoint too far). Since the flight mode changed, it then happily armed the vehicle as well.

As I said above, I already tried reversing the order and it made no difference.

DonLakeFlyer, 2017-07-08:
So I'm confused now: @LorenzMeier says he has the proper fix which are far as I can tell doesn't really work (at least for what I tested) and also doesn't use MISSION_START. Should I switch to using MISSION_START or not?

LorenzMeier, 2017-07-08:
I tested it and it does work. Can you please re-test on your side?

DonLakeFlyer, 2017-07-08:
Yup, will do.

LorenzMeier, 2017-07-08:
Mission reject before acquiring GPS lock:
[image]

Mission flown after acquiring lock:

[image]

DonLakeFlyer, 2017-07-08:
No GPS lock: Fails switch to flight mode
Mission too far away: Allows switch to Mission flight mode

Either way, not failing Mission flight mode is firmware side problem. So when firmware does fail switch, it does not arm. So QGC side is working correctly. Just doesn't seem like all causes are handled correctly on firmware side for flight mode change.

Antiheavy, 2017-07-08:
I'd like to clarify that the original issue I posted above had to do specifically with PX4 "FW Launch Detection" that is designed for catapult (or hand launched in our case) fixed wings.  The original issue is that the system seems to allow the aircraft to switch to mission mode and arm, but "FW Launch Detection" does not kick in preventing the "land detector" from starting the motor(s) when the aircraft is moved.

Maybe this problem is related to everything else discussed here (?), but 1) correctly preventing arming for various valid reasons and 2) correctly allowing arming, but failing to start "FW Launch detection" are two separate issues from the users perspective.

DonLakeFlyer, 2017-07-08:
That feels like a separate problem from what is being discussed here. I'd post a firmware issue with vehicle log for repro so firmware folks can look.

dagar, 2017-07-08:
@Antiheavy It's possibly also a firmware issue, but in at least one of your logs the issue was arming, but not going into mission mode which skips over the launch detect.

LorenzMeier, 2017-07-09:
This should be fixed now here: https://github.com/PX4/Firmware/pull/7564 - please reopen in Firmware if it does persist.

closed 2017-07-09 by LorenzMeier