mavlink/qgroundcontrol #5390: Mission planning feedback:

AndKe, 2017-07-03:
While planning a survey of an new area, it's very useful to fly out, to verify visually it's limits.
Please:
1:  just like in "fly view" draw a red line when flying in planning mode, (let user verify/fly around the area from the air, then look on the laptop, and move perimeter accordingly, also it's significantly easier if the area is a more complex polygon, not just a four corner box, )
2:  display video feed, list like in fly view.  
3: display main flights instruments, especially battery voltage.

DonLakeFlyer, 2017-07-03:
Interesting, but not really feasible. Would cause all sort of problems. Seems like this could partially be accomplished by just using the Fly view. Send your plan to the vehicle so you can see it in the fly view. Not just fly the vehicle manually to wherever you want.

AndKe, 2017-07-04:
Ok, so I'll upload the plan, let's imagine I fly over the plan, and observe it's too wide in one direction, I move the point, now I fly further, and another point needs to be moved, so I need to upload the mission as the polygon changes, - not having track of previous path, maybe I have shrunken the area somewhere in the middle.

Is it that hard to draw the red path in the mission planning ?  - if so, the video feed would at least reduce the amount of jumps between, and so provide a faster edit with uploading.

(it's not really only the outer perimeter that needs to be placed, but the different altitudes for different parts of the survey).

closed 2026-05-16 by AndKe