mavlink/qgroundcontrol #5392: Please add progress for mission/params on connection.

AndKe, 2017-07-03:
After connecting to a UAV, in fly view, one just sit there, waiting for the mission to spawn.
It would be very nice to have a progress bar (like the upload bar in planning view)  that indicated parameter, and mission download progress.

DonLakeFlyer, 2017-07-03:
I'm working no new mission loading archtiecture right now. I'll include that as part of it.

DonLakeFlyer, 2017-07-07:
I didn't end up pushing my re-arch in 3.2 so this is going to have to wait.

SaliniVenate, 2017-07-12:
@DonLakeFlyer 

> I'm working no new mission loading archtiecture right now.

Is this mission loading architecture proposal published anywhere, that I can refer to?

DonLakeFlyer, 2017-07-15:
Nothing written up. Just a pull I was working on. Ended up bailing out on it for 3.2.