mavlink/qgroundcontrol #5393: Mission resuming: modification, logic, and crazy result.

AndKe, 2017-07-03:
1: I assume you already observed that the "refly at 90degree" survey option is prone to producing long transport distance, rather than resuming at 90degree nearby where the previous direction ended.

2: Once I land, there is an "slide-to" option to recalculate the mission (basically crop it)  - i did not prefer to prefer to use it, but rather have the "whole view" of progress of multi-level flights all the way.
- also , I just wanted to swap battery and restart, this is Solo, so I have no groundpower, for it, and did not want a upload process every time)

3: Once I were uploading a mission, and swapped battery - the sync button were stuck later on.
 
4: So I did not respond to the offer to "trim-down mission" after swapping battery.  Once I took off, I were asked to resume mission - which always went to item number one".   Please, make resume, actually resume, even when not trimming down mission (set correct WP after reboot, on or before takeoff)

5: Once and the end of day2 I decided to give the resume "mission trimming" a chance, the result is below:

[image]

apparently it removed/messed up all camera commands, as no photos were trigged, I did not try to figure it out, just uploaded original mission(attached), and jumped to correct wp like before.

I do not see a great use for modifying missions rather than just doing a jump.

6: please, if user uses "pause" , or "RTL" , or switch mode away from AUTO, do store last WP(unless mission is completed), and push it out when the vehicle boots up again (so resume actually works in every scenario)

[Repparfjord.plan.zip](https://github.com/mavlink/qgroundcontrol/files/1120363/R
epparfjord.plan.zip)
[image]

DonLakeFlyer, 2017-07-03:
> I assume you already observed that the "refly at 90degree" survey option is prone to producing long transport distance, rather than resuming at 90degree nearby where the previous direction ended.

That should not be the case. Does the attached show that behavior.

> I do not see a great use for modifying missions rather than just doing a jump.

Simple example. You slow the vehicle down below the default mission cruise speed to perform the survey. If you just jump into the middle of the mission it is going to miss the change of flight speed and fly the rest of the survey at the wrong speed.

AndKe, 2017-07-04:
Sorry, I mean literally "modifying the mission" , as is, uploading it as another set of waypoints.
I do like, and understand the need to parse the mission, and issue the essential changes as a part of actual resume, before executing the waypoint at which to resume.

Instead of re-uploading another  mission with hundreds of items, just fast forward it on the GCS, look for final speed/camera angle/ exposure time , or other settings, issue those as mavlink commands , and verify those. (in case of bad link)

The advantage using such method is; no huge upload over slower telemetry (re-upload is a no-no if using Iriduim as primary link) , Then we have trace ability in case some logs need reviewing, wp number 422  would always be 422, not become  150 then 80.   analyzing the final part, would still be perfectly readable when comparing to the mission file.

Finally, it's easier to be sure it works as expected, once speed/options are reset, and mission continues from wp 100 , one can rely on it be as planned.  (no hiccup during processing or upload)

are there any advantages uploading a modified mission ? 
if so, maybe the best option would to offer both methods ?

As for the exit vs entry point when switching between directions "doing 90degree",  yes, I have that in one of the missions from the past days,  will upload and check.  This reminds me of my previous request to be able to "explode" and review a mission, or export to QGC WPL file (as that could be imported and verified easily.)

AndKe, 2017-07-04:
I checked the missions, my point "1" , the diagonal survey travel is not an issue¸ I failed to observe that the diagonal travel I saw were between two surveys at different altitude, not two direction of the same survey.  (sorry)

DonLakeFlyer, 2017-07-07:
> Instead of re-uploading another mission with hundreds of items, just fast forward it on the GCS, look for final speed/camera angle/ exposure time , or other settings, issue those as mavlink commands , and verify those. (in case of bad link)

Ok, I see what you are saying. That would only work once since you lose the re-start commands from the mission. If you needed another battery you wouldn't be able to resume since the important starti[ commands which not be part of the mission.

DonLakeFlyer, 2017-07-07:
> That would only work once since you lose the re-start commands from the mission.

That is an incorrect statement.