mavlink/qgroundcontrol #5394: Alert popup, and suppression of known messages:

AndKe, 2017-07-03:
-Please observe that the yellow popup is often way behind.
-Please suppress known , useless messages like the Arm: OK from Solo (or maybe that should be solved on the Solo firmware side) 
.
[image]

DonLakeFlyer, 2017-07-03:
> Please observe that the yellow popup is often way behind.

There is an issue for a redesign of that in 3.3

The other message stuff is strange. I've never gotten that when I fly mine. Do you some strange param set of something which makes this stuff spit out?

AndKe, 2017-07-03:
no, I have a gimbal with RX100, some changed PID's but nothing special that should cause that .
That said, I also have a stock Solo i fly now and then, and Solex says "high message rate" or something like that, author said that happened when solo sent much statustext - could be related.   (QGC was the only GCS these days, so I am just mentioning it, but must point out I did not use Solex on those missions)

vidma, 2022-03-20:
guys this is super annoying. could we have at least an option to disable all such popups!!!???

DonLakeFlyer, 2022-03-20:
> Please suppress known , useless messages like the Arm: OK from Solo (or maybe that should be solved on the Solo firmware side)

There is no way for QGC to know which errors are real and which are not. In reality this is a firmware problem since it is spamming errors to the vehicle. QGC side changes are just workarounds for bad firmware.

AndKe, 2022-03-20:
@DonLakeFlyer  it's not all about bad firmware.  sometimes it's payload,  even current plane/copter releases when using NMEA Yaw source indicate that  they have "found a NMEA device "  (other wording) every x seconds. - is it a bug/undesired behavior - yes, but it is still a good idea to have a less intrusive way to display such messages.  

On MAVProxy - they are confined to the console  module, in AP2 - they are in own status-window.   - only QGC allow them to fill the screen/cver working area.
One suggestion would be to be able to detach that as a separate window - and move it out of the way.

vidma, 2022-03-21:
> One suggestion would be to be able to detach that as a separate window - and move it out of the way.

i like that. could be even simplified for now IMHO - simple option/button to silence (disable all popups)  would be enough, as there's a separate window to view messages already.  maybe a standard loglevel filter might also help, but I don't think that's so important (?)

DonLakeFlyer, 2022-03-21:
> but it is still a good idea to have a less intrusive way to display such messages.

They are meant to be intrusive on purpose. There are multiple levels of severity associated with mavlink status text message. The highest severity is error, which indicates something bad is wrong on the vehicle side. Only this highest level of severity is shown in a popup. All other messages are relegated to the dropdown. If something bad is happening in the vehicle you want to know about it right now. I can't help the fact the whatever is on the vehicle side is sending crap with incorrect severity. There is no way for QGC to know which of these messages are real and you want to know about them right away and which are not. You should not be able to hide this or move something out of the way that could possibly indicate your vehicle is in trouble.

DonLakeFlyer, 2022-03-21:
Or in other words if the fire alarm in your house is going off inadvertently you don't pull the battery to silence it you replace the faulty fire alarm.

AndKe, 2022-03-21:
@DonLakeFlyer - The fire alarm does not obstruct your way to the fire extinguisher, then does it again. or does not only make sound in one specific room.

Repeating popups could hinder the GCS pilot's ability to use guided mode effectively.

Also - the alarms are only visible in main flight window, not vehicle setup or analysis mode. (I would say the information you describe as critical, should be available from anywhere)

It would satisfy all to have the option to move it out of the way, most have enough screen estate or external displays to gladly utilize them - and this is a great thing to use them for.  I would certainly like to know if even an medium severity event like EKF track change started to repeat, - regardless of which part of the GCS I am looking at.

DonLakeFlyer, 2022-03-21:
> Repeating popups could hinder the GCS pilot's ability to use guided mode effectively.

Dealing with just that part is what I was talking about with respect to redesign. Just the repeat part. Never got around to it though. And now my QGC time is next to non-existent. Wish DroneCode would figure out how to get someone to be the new maintainer to deal with stuff like this.

DonLakeFlyer, 2022-03-21:
> It would satisfy all to have the option to move it out of the way, most have enough screen estate or external displays to gladly utilize them - and this is a great thing to use them for. I would [...]

All makes sense.

AndKe, 2022-03-21:
> Wish DroneCode would figure out how to get someone to be the new maintainer to deal with stuff like this.
@DonLakeFlyer 

I admire all the work you've put in into QGC - the only updated, graphical, GCS worthy of ArduPilot/PX4 in 2022 (being multiplatform)  - and I did not even know that DroneCode was still a thing.  - kind of ignorant of me I guess.

closed 2026-10-05 by github-actions[bot]