mavlink/qgroundcontrol #5397: Never pan map to lat/lon ~0.000 / 0.000

AndKe, 2017-07-04:
Reproduce:
Start application
Wait till an UAV with mission onboard connnects.
Once mission is downloaded, you will briefly fee it in the map, before map pan to UAV that does not have GPS fix, in the middle of nowhere.   Should operator wish to verify/see mission, then he needs to zoom out and do so manually:

[image]

DonLakeFlyer, 2017-07-06:
Sounds like more cases of ArduPilot making me crazy by sending crap gps values! I thought I had cauterzied all of these.

DonLakeFlyer, 2017-07-15:
This is a 0/0 lat/lon coming through from first fake home position in ArduPilot.

closed 2017-07-17 by DonLakeFlyer