mavlink/qgroundcontrol #5398: Plan view: please allow clicking on the lower mission items

AndKe, 2017-07-04:
It would be nice if clicking on the lower set of mission items highlighted the correct mission item, or - behaved as if the one in the map were clicked. (that would also workaround the "Z-problem")

[image]

DonLakeFlyer, 2017-07-06:
Hmm, I'll have to look. I thought that was already working. Maybe it broke somewhere along the way.

DonLakeFlyer, 2017-07-07:
No it doesn't work. I'll make it work in 3.3. For now you can click on something in the map to get close and then scroll the right hand mission item list to get to way you want and click there.

DonLakeFlyer, 2020-11-20:
Supported now

closed 2020-11-20 by DonLakeFlyer