mavlink/qgroundcontrol #5400: qGroundControl Autonomous Rover

bwheatl, 2017-07-05:
I am using a Pixhawk Mini with a qGroundcontrol rover. I can start qGroundcontrol and use my Spektrum radio to control the rover. I tried using autonomous control by setting up waypoints. When I tell it to execute nothing happpens. I also noticed I can only control the rover using the radio if qGroundControl is in manual mode. Are these just limitations of qGroundcontrol with a rover or is something not setup properly?

dagar, 2017-07-05:
Are you using PX4 or ArduRover?

bwheatl, 2017-07-05:
Yes. I'm sorry I forgot to mention the PX4.

dagar, 2017-07-06:
Rover support in PX4 is new and has only been tested with this setup. https://dev.px4.io/en/airframes_experimental/traxxas_stampede.html

I would go through that and make sure you have everything configured. If you find problems please open issues at the PX4 Firmware repo (https://github.com/PX4/Firmware) or post on PX4 discuss (http://discuss.px4.io/). It's usually helpful to share a log (http://logs.px4.io/) along with a description when you hit a problem.

Closing this as it's not a QGC issue.

closed 2017-07-06 by dagar