mavlink/qgroundcontrol #5403: Mission finished remove question optional

sanderux, 2017-07-06:
@DonLakeFlyer 
Could you make the new "Mission finished, remove?" question optional?

DonLakeFlyer, 2017-07-06:
My opinion is that nobody ever finds options and that adding more of them just makes things more confusing. Do you have a reason why removing a mission after flying it is not the majority use case?

sanderux, 2017-07-06:
I have several:
1. Missions can now be flown in simulation before sending them to the drone.
2. Ekf replay sometimes needs the mission on the dataman file to replay.
3. Repetative missions like an oil pipeline inspection could require the same mission being flown over and over again.
4. The mission might be controlled from the onboard computer making this just another click.

How about a 'dont show again' checkbox?

dagar, 2017-07-06:
Why is it important to actually remove the mission from the vehicle? Isn't it pretty harmless to leave in the finished state?

sanderux, 2017-07-06:
And a user might have read only access to the drone

Antiheavy, 2017-10-05:
Coming late to the party:  I think it would be useful to have a firmware param to optionally "erase old missions on boot".  My my experience users (customers) sometimes accidentally try to fly old missions that might be at a different location, this can be very dangerous if the built-in checks don't catch a problem with the mission.

I'm not a huge fan of the "Mission finished, remove?" question, however, it partially serves a purpose to remind the user the should have a "clean" autopilot before going to their next location or next flight plan.  In this sense the function as currently implemented saves the user a bunch of clicks.

closed 2026-10-04 by github-actions[bot]