mavlink/qgroundcontrol #5404: Mission speed displayed differently

ChristophTobler, 2017-07-06:
When changing the the flight speed in mission start the whole mission will have this flight speed (see flight plot) but the it is not displayed this way (current master)
[image]
[image]
[image]

So changing/updating the displayed value for future waypoints fixes this.

DonLakeFlyer, 2017-07-06:
Are you talking about the grayed out flight speed values in subsequent waypoint items? Those are just the default flight speed. Not meant to  indicate the flight speed you may have set before that waypoint. But still if that's it, it is confusing and would be better if it showed the previous flight speed set.

ChristophTobler, 2017-07-07:
Yup, that's what I meant.

DonLakeFlyer, 2017-12-30:
Fixed

closed 2017-12-30 by DonLakeFlyer

estherpolie, 2024-04-30:
hi Don , where is the source code for flight speed