mavlink/qgroundcontrol #5416: PX4 unused parameters MPC_XY_FF MPC_Z_FF

dagar, 2017-07-08:
In PX4 I recently deleted the parameters MPC_XY_FF and MPC_Z_FF because they weren't actually connected to anything in the controller. It looks like QGC is trying to use them on the tuning page.

[image]

DonLakeFlyer, 2017-07-08:
So does this capability not exist:
```
                ListElement {
                    title:          qsTr("Altitude control sensitivity")
                    description:    qsTr("Slide to the left to make altitude control smoother and less twitchy. Slide to the right to make altitude control more accurate and more aggressive.")
                    param:          "MPC_Z_FF"
                    min:            0
                    max:            1.0
                    step:           0.1
                }

                ListElement {
                    title:          qsTr("Position control sensitivity")
                    description:    qsTr("Slide to the left to make flight in position control mode smoother and less twitchy. Slide to the right to make position control more accurate and more aggressive.")
                    param:          "MPC_XY_FF"
                    min:            0
                    max:            1.0
                    step:           0.1
                }
```

dagar, 2017-07-08:
You can use MPC_Z_P and MPC_XY_P.

@Stifael @MaEtUgR FYI

DonLakeFlyer, 2017-07-08:
Ok, thanks Daniel.

DonLakeFlyer, 2017-07-08:
Is this:
```
                    min:            0
                    max:            1.0
                    step:           0.1
```
Correct for these new params as well?

dagar, 2017-07-08:
Can it be set from param metadata? If not here's the current values.

MPC_XY_P max is 2.0.
MPC_ZZ_P max is 1.5.

DonLakeFlyer, 2017-07-08:
If I remember correctly the tuning sliders min/max can be different than the absolute min/max from meta data. Also the increment meta data isn't always there. So I put these in there by hand. For example neither MPC_XY_P nor MPC_Z_P have increment meta data specified so not sure what to set "step" to.

dagar, 2017-07-08:
Oh, I assumed this was exactly what the metadata was for. Should we do a pass and fill in all required metadata on the PX4 side?
It might also be useful to have a list of all params QGC cares about to fix any other issues like this.

DonLakeFlyer, 2017-07-08:
The concept is that for simple uneducated user tuning ui you may not want to pull in min/max since the full range may make things a little crazy. Can't remember if that is actually used but that was the idea. If not it could use meta data directly.

For now since this ui control doesn't reference the param meta data it will be easier to just stuff them in manually. I'll put in an issue to revisit this to plumb directly to meta data.

dagar, 2017-07-08:
We should really try to resurrect this at some point. https://github.com/mavlink/qgroundcontrol/pull/5070

DonLakeFlyer, 2017-07-08:
Yeah I want to make a push on vehicle tuning, analysis as a focus for 3.3.

dagar, 2017-07-09:
Great! I still use the current analyze in all its pastel colored glory several times/week.

DonLakeFlyer, 2017-07-09:
Fix by Lorenz

closed 2017-07-09 by DonLakeFlyer