mavlink/qgroundcontrol #5432: Missing topics in analyzer

ChristophTobler, 2017-07-13:
I pulled master and now I don't see all the topics anymore in the analyzer
[image]
It still worked with https://github.com/mavlink/qgroundcontrol/commit/b682031e7b17ec10a92fc65af29f6bbf93939ba7

jgoppert, 2017-07-14:
I can confirm.

DonLakeFlyer, 2017-07-15:
Most likely broken by this: https://github.com/mavlink/qgroundcontrol/pull/5420

DonLakeFlyer, 2017-07-15:
I just tried QGC master against px4 firmware master SITL using both a V1 and V2 connection and it all worked fine?

DonLakeFlyer, 2017-07-15:
Ok, master seems to work for me but Stable 3.2 does not. Hmm?

DonLakeFlyer, 2017-07-15:
This is in master and stable 3.2 now, but will require point release on Stable. Will do that in a few days to collect other fixes.

ChristophTobler, 2017-07-17:
Works in master. Thanks!

closed 2017-07-17 by ChristophTobler