mavlink/qgroundcontrol #5434: Issue with Geofence

SaliniVenate, 2017-07-14:
I'm connecting QGC to jMAVSim SITL running px4 (version 1.6.5dev). I'm not able to set a geofence for a mission.Here is the steps:
1)I selected the geofence button in Plan view and entered the breach action, radius and altitude.
2) Send the mission to vehicle by clicking on "upload" in plan tool "sync"
Thanks

SaliniVenate, 2017-07-14:
@dagar @DonLakeFlyer

dagar, 2017-07-14:
@SaliniVenate this geofence editing api hasn't been merged in PX4 yet. 
Watch this issue - https://github.com/PX4/Firmware/pull/7238

Until then you can create a simple cylindrical geofence in QGC settings, or a polygon fence on the sdcard in the autopilot.

@bkueng @DonLakeFlyer do we have a mechanism to check if the autopilot supports this? At a minimum can we hide it in QGC, perhaps with a setting to enable?

bkueng, 2017-07-14:
I don't know why QGC already shows an editing UI, the relevant PR is still WIP (https://github.com/mavlink/qgroundcontrol/pull/5147).
In any case there are capability bits for the new API that QGC can check.

DonLakeFlyer, 2017-07-15:
> Until then you can create a simple cylindrical geofence in QGC settings

As Daniel said. This is all working correctly. Current QGC does not support polygon geofence.

SaliniVenate, 2017-07-17:
@dagar, So as of now with QGC, to set a cylindrical geofence and  I've to check "Geofence Failsafe Trigger" in "Safety" in the vehicle setup and to remove a geofence I've to un-check that. I cannot set/remove a geofence from plan view right?

DonLakeFlyer, 2017-07-17:
You should also be able to remove a GeoFence from the Plan view as well. let me take a look.

DonLakeFlyer, 2017-07-18:
Setting PX4 GeoFence from plan view is not working correctly. You can set it up from the Safety page where it does work correctly.

DonLakeFlyer, 2017-07-18:
Latest PR fixes the ability to set fence action/radius/height from plan view. Remove All still does not work though.

closed 2017-12-30 by DonLakeFlyer