mavlink/qgroundcontrol #5435: Maps doesn't load in Version 3.2

diegoeck, 2017-07-14:
The maps doesn't load after I upgraded to Version 3.2. Using Version 3.1.3 there is no problem.

It something related to Internet Access. The problem happens with both AppImage version and Compressed Version. Ubuntu 16.04 LTS.

Here is the console log:

diego@diego-ThinkCentre-M93p:~/Downloads/qgroundcontrol$ ./qgroundcontrol-start.sh 
Settings location "/home/diego/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/diego/Documents/QGroundControl/Logs"
Map Cache in: "/home/diego/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: 0.037109375
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: 34.5 38
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
Orphaned log file count 0
No Internet Access
QGeoTileRequestManager: Failed to fetch tile (2,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,1,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,2,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (2,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (1,0,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (3,3,2) 5 times, giving up. Last error message was: 'Network not available'
QGeoTileRequestManager: Failed to fetch tile (0,0,2) 5 times, giving up. Last error message was: 'Network not available'
No Internet Access

dagar, 2017-07-14:
Building from source and daily AppImage are both working for me on Ubuntu 16.04. Could it be some other dependency I already have on my machine? Full Qt dev environment installed.

NaterGator, 2017-07-14:
@diegoeck can you ping 8.8.8.8? Does something in your network topology block DNS access to outside servers? 

Seems most likely this was introduced by [this](https://github.com/mavlink/qgroundcontrol/commit/23cb5e1b950528e717d701446cb736b0e114e8a3) commit. @dogmaphobic maybe instead of connecting to Google DNS we should try opening port 80 to the domain of the selected map provider?

dogmaphobic, 2017-07-14:
>maybe instead of connecting to Google DNS we should try opening port 80 to the domain of the selected map provider?

That would require two connections. A DNS query to get the map provider's IP address and a subsequent connection to the map server. Even though that code is on a separate thread, the DNS query (`gethostbyname`) would block the thread for too long if you don't have Internet access.

My assumption was that if you can't reach 8.8.8.8, you can't reach anything.

diegoeck, 2017-07-14:
I can try ping 8.8.8.8 monday.
We have some "complicated" network infrastructure at my university, but the previous version is working.

I believe that we can reach Google DNS, but I need to confirm.

diegoeck, 2017-07-17:
I can ping 8.8.8.8, it is not the problem.

diego@diego-ThinkCentre-M93p:~$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=43 time=70.9 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=43 time=70.7 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=43 time=69.7 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=43 time=69.8 ms
64 bytes from 8.8.8.8: icmp_seq=5 ttl=43 time=69.5 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4007ms
rtt min/avg/max/mdev = 69.597/70.195/70.959/0.611 ms

dogmaphobic, 2017-07-19:
The question is what makes your setup unique? Is this happening to anyone else? I need to know what the source of the problem is before I can find a way around it.

DonLakeFlyer, 2017-07-19:
I'm moving this to 3.3 since it seems like very network setup specific. If that is not the case and it is happening to many people, move back to 3.2

diegoeck, 2017-07-19:
What other information should I provide? If it is a network problem, it should not work with other OS. I will try tomorrow with macOS and Android.

Baycken, 2017-07-21:
I am running into the same issue in both Linux and MacOS with online map. 
I can still download offline map.

Baycken, 2017-07-22:
I did some more testing today.It looks like the problem is network specific. 

I installed version 3.2.1 on MacOS and failed to get map within the my university's network. The same machine is brought back to my house and the maps work with the internet that I have at home.

Ganser7342, 2017-07-29:
I would like to add that I am having the same issue with QGC v3.2.1 on both a Mac running OS X 10.12 and a Windows 7 machine. The internet is connected via WIFI (802.11g) on a multi access point, home network. I did tether my phone's LTE service to the Mac and was able to get the maps working. Please let me know if I can add further specifications of the interface to help get this resolved.

DonLakeFlyer, 2017-07-29:
@dogmaphobic Can you figure out some way to debug this?

DonLakeFlyer, 2017-07-29:
Also moved back to 3.2 since it seems to not just be a fluke with strange network setup.

dogmaphobic, 2017-07-30:
I'm assuming we have only a couple of reports on this? That is, for most everyone else it all works fine? What is common with the people who reported it? I can't remember any changes to the code related to fetching tiles. Does it happen with all map providers? It seems it affects different OSs. The only common factor is the router. It must be some router configuration. Proxy? Firewall?

mad-sci, 2017-07-30:
Having the same issue. Map loads in 3.1.3 but not in 3.2 on windows 8.1machine.

Ganser7342, 2017-07-30:
@dogmaphobic I am getting no maps on both OSs, on a low security level home network. It also dosen't matter what map provider is selected. The latest version of Mission Planner brings the maps up fine on both OS's.

NaterGator, 2017-07-30:
> My assumption was that if you can't reach 8.8.8.8, you can't reach anything.

That's probably a pretty reasonable assumption, except traditional DNS queries are UDP based (and we could pick something like google.com, which the OS is likely to have cached locally anyways). My thought here is some networks may be blocking TCP traffic on DNS ports for one reason or another.

NaterGator, 2017-07-30:
Windows users with this problem, can you use [this](https://ci.appveyor.com/api/buildjobs/mp6j8yjnp3225r3f/artifacts/QGroundControl-installer.exe) version and check the "Bypass network connectivity checks" box under General settings to verify there isn't something else going on with your network:
[image]

Other platforms feel free to build https://github.com/NaterGator/qgroundcontrol/commit/c53ecd5ab1b77515b92d393b25c41ad90b4b07bc but I don't there there is a way to get the installer artifacts from the CI server.

DonLakeFlyer, 2017-07-30:
> I'm assuming we have only a couple of reports on this?

Various Sentera folks have reported the same thing using their Windows version. So the count of people with problems keeps going up.

DonLakeFlyer, 2017-07-31:
Another check is to just look for the log output "No Internet Access" in the Console log. That will tell you if the internet check for map tiles is failing. If you can't run Nate's build then please let us know what is in you console log output. You should just need to boot and look at Setup/Console. Here is an example from my machine unplugged from internet:

[image]

Ganser7342, 2017-08-01:
@NaterGator I went ahead and ran your build of QGC, checked the "Bypass network connectivity checks" box, and now it's working.  This is on the Windows 7 desktop. Before hand I went ahead and turned off the firewall, but it still wasn't working. Your latest build did the trick.

NaterGator, 2017-08-02:
@dogmaphobic what do you think about a simple get request to http://www.gstatic.com/generate_204 and looking for an HTTP 204 status code?

dogmaphobic, 2017-08-02:
Anything that works world over is fine. Even github.com:80 should work. We don't need to *GET* anything. Meaning, in you example, it would be a simple TCP connection on port 80 to `www.gstatic.com`. As soon as the connection succeeds, we know we have Internet.

I do wonder why all of the sudden routers started blocking TCP to port 53. Minimize DDOS attacks?

DonLakeFlyer, 2017-08-02:
You could always pull something from our S3 servers. Also just FYI, the fix for this needs to go into stable.

dogmaphobic, 2017-08-02:
@NaterGator I will push a "fix" for this in a bit.
@DonLakeFlyer I will create a branch for it and merged to both stable and master.

dogmaphobic, 2017-08-02:
Come to think of it, the reason I was connecting to 8.8.8.8 was to avoid a DNS lookup, which can lock the thread for quite a bit if you don't have access. Is there another known IP address that could be used?

dogmaphobic, 2017-08-02:
It's here: https://github.com/mavlink/qgroundcontrol/pull/5520

DonLakeFlyer, 2017-08-03:
Thanks everyone for this one. This is a great find/fix to get in.

dogmaphobic, 2017-08-12:
This is back to hunt us. The reason I had an IP address was to avoid a DNS lookup. Now that we're looking for www.github.com (or anything else for that matter), the code goes through a DNS lookup. On Android, with no Internet connection, this takes 60 seconds before it gives up. If you try to exit QGC, the lookup thread is blocked for up to 60 seconds before it closes.

DonLakeFlyer, 2017-08-12:
https://stackoverflow.com/questions/4238921/detect-whether-there-is-an-internet-connection-available-on-android?

dogmaphobic, 2017-08-12:
That only detects if you have a network connection, which you do. I've worked around by splitting the host lookup from the connection. The host lookup can be aborted.

DonLakeFlyer, 2017-08-12:
Ok, can you make your pull against Stable_V3.2? Then I'll merge to master and you can pull to where you need. It's much easier to make stable changes in the order.

DonLakeFlyer, 2017-08-17:
This is all fixed now. Just waiting for new round of 3.2 builds to kick off.

closed 2017-12-29 by DonLakeFlyer