mavlink/qgroundcontrol #5436: Recorded video stream is not playable

dogmaphobic, 2017-07-14:
There is still some issue with the recorded video stream as a lot of video players can't play them, including Qt itself. QuickTime player won't play but VLC will play them though.

@jaxxzer Any clues?

I used this snipped to try it. It can play any other .mp4 file. It just won't play the ones recorded by QGC.

```
Video {
    width:          1280
    height:         720
    source:         "/path/to/recorded/video.mp4"
    autoLoad:       true
    onErrorStringChanged: {
        console.log(videoContent.error + ' ' + videoContent.errorString)
    }
}
```

jaxxzer, 2017-07-18:
Is .mp4 the only one you are having trouble with? Do you have a qgc branch or project you can push so I can use that?

NaterGator, 2017-07-18:
Any samples of videos that fail in most players?

dogmaphobic, 2017-07-18:
>Is .mp4 the only one you are having trouble with?

No, it's not the container. The other containers have the same issue. Well, .mov and .mp4. VLC is the only thing that will play .mkv and VLC seems to handle the problem video just fine so I can't tell.

Try this: http://www.grubba.com/2017-07-14_17.56.55.mp4 (not the most interesting video but...)

NaterGator, 2017-07-18:
Looks like the decode timestamps are bogus:

```C:\Users\nweib\Videos>ffmpeg -i 2017-07-14_17.56.55.mp4 -f null -
ffmpeg version N-86789-g4c1aac8 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 7.1.0 (GCC)
  configuration: --disable-static --enable-shared --enable-gpl --enable-version3 --enable-cuda --enable-cuvid --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-nvenc --enable-avisynth --enable-bzlib --enable-fontconfig --enable-frei0r --enable-gnutls --enable-iconv --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libfreetype --enable-libgme --enable-libgsm --enable-libilbc --enable-libmodplug --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenh264 --enable-libopenjpeg --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs --enable-libxvid --enable-libzimg --enable-lzma --enable-zlib
  libavutil      55. 67.100 / 55. 67.100
  libavcodec     57.100.104 / 57.100.104
  libavformat    57. 75.100 / 57. 75.100
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 95.100 /  6. 95.100
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
  libpostproc    54.  6.100 / 54.  6.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '2017-07-14_17.56.55.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41isomiso2
    creation_time   : 2017-07-14T21:56:55.000000Z
  Duration: 00:00:13.25, start: 0.091900, bitrate: 2032 kb/s
    Stream #0:0(eng): Video: h264 (Extended) (avc1 / 0x31637661), yuv420p, 1376x720, 2026 kb/s, 30.04 fps, 10k tbr, 10k tbn, 20k tbc (default)
    Metadata:
      creation_time   : 2017-07-14T21:56:55.000000Z
      handler_name    : VideoHandler
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41isomiso2
    encoder         : Lavf57.75.100
    Stream #0:0(eng): Video: wrapped_avframe, yuv420p, 1376x720, q=2-31, 200 kb/s, 30.04 fps, 30.04 tbn, 30.04 tbc (default)
    Metadata:
      creation_time   : 2017-07-14T21:56:55.000000Z
      handler_name    : VideoHandler
      encoder         : Lavc57.100.104 wrapped_avframe
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 6 >= 6
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 13 >= 13
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 31 >= 31
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 44 >= 44
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 50 >= 50
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 53 >= 53
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 55 >= 55
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 62 >= 62
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 65 >= 65
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 79 >= 79
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 81 >= 81
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 85 >= 85
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 88 >= 88
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 113 >= 113
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 140 >= 140
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 142 >= 142
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 165 >= 165
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 167 >= 167
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 172 >= 172
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 174 >= 174
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 176 >= 176
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 178 >= 178
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 199 >= 199
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 202 >= 202
[... cut ...]
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 391 >= 391
[null @ 0000000000eecd20] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 395 >= 395
frame=  392 fps=0.0 q=-0.0 Lsize=N/A time=00:00:13.18 bitrate=N/A speed=78.3x
video:202kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown```

It may be time to audit the clock distribution in the gstreamer pipeline. I'll poke around some more but I wouldn't be at all surprised to see inconsistent DTS/PTS tripping up players.

DonLakeFlyer, 2017-12-29:
Is this still a problem?

dogmaphobic, 2017-12-29:
Yup, very much so. I have no idea what the problem is.

jaxxzer, 2018-01-02:
I've dug into it a little bit, it's an issue with timestamping as @NaterGator pointed out. I have not resolved it, but it is on my list.

dogmaphobic, 2018-01-03:
I've been running this to test:

```
gst-launch-1.0 rtspsrc location="rtsp://192.168.42.1/live" ! rtph264depay ! h264parse ! tee name=t t. ! queue ! avdec_h264 ! autovideosink t. ! queue ! matroskamux ! filesink location="video.mkv
```

The resulting file will play on Android but not on macOS (I have not tested other platforms)

For what's worth, `ffprobe` has this to say about it:

```
ffprobe E50.mkv
Truncating packet of size 21863284 to 1129486
[h264 @ 0x7fb50480f000] missing picture in access unit with size 357
[h264 @ 0x7fb50480f000] no frame!
Input #0, matroska,webm, from 'video.mkv':
  Metadata:
    encoder         : GStreamer matroskamux version 1.10.2
    creation_time   : 2018-01-02T20:35:20.000000Z
  Duration: N/A, start: 0.050000, bitrate: N/A
    Stream #0:0(eng): Video: h264 (Baseline), yuvj420p(pc, bt709, progressive), 848x480 [SAR 1:1 DAR 53:30], 30 fps, 30 tbr, 1k tbn, 60 tbc (default)
    Metadata:
      title           : Video
```

If instead of Matroska I switch the muxer to m4p or mov (quicktime), it doesn't play anywhere.

jaxxzer, 2018-01-10:
> If instead of Matroska I switch the muxer to m4p or mov (quicktime), it doesn't play anywhere.

I think this is because you do not have -e to push EOS so that mp4mux can write the header needed by mp4 container.

dogmaphobic, 2018-01-10:
What is "-e"? How do you define that within the (c++) pipeline?

jaxxzer, 2018-01-10:
-e is an option on gst-launch to push END OF STREAM on the pipeline after ctrl-c. We do that in our application [here](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/VideoReceiver.cc#L776).

dogmaphobic, 2018-01-10:
That's what I thought, but the video recorded within QGC has the same issue:

```
$ ffprobe E90.mkv
Truncating packet of size 21863284 to 2756261
Input #0, matroska,webm, from 'E90.mkv':
  Metadata:
    encoder         : GStreamer matroskamux version 1.10.2
    creation_time   : 2018-01-02T20:32:53.000000Z
  Duration: N/A, start: 0.046000, bitrate: N/A
    Stream #0:0(eng): Video: h264 (Extended), yuv420p(progressive), 1280x720, SAR 1:1 DAR 16:9, 1k tbr, 1k tbn, 2k tbc (default)
    Metadata:
      title           : Video
```

In addition, the time stamp (when playing) is all over the place.

dogmaphobic, 2018-01-10:
I deleted the entire comment because it was wrong. I was testing the wrong file. Here is the result of `ffprobe` on a file recorded within QGC:

```
$ ffprobe 2018-01-02_22.34.11.mkv
Input #0, matroska,webm, from '2018-01-02_22.34.11.mkv':
  Metadata:
    encoder         : GStreamer matroskamux version 1.5.2
    creation_time   : 2018-01-03T03:34:11.000000Z
  Duration: 00:00:20.29, start: 68.672000, bitrate: 986 kb/s
    Stream #0:0(eng): Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1280x720 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 1k tbn, 59.94 tbc (default)
    Metadata:
      title           : Video
```

jaxxzer, 2018-01-16:
Here's my observations:

I used this command to reproduce @NaterGator's results:

`ffmpeg -stats -v debug -i test.mkv -f null -`

The ffmpeg error about non-monotonically increasing timestamps appears to be an issue with gstreamer in general, but does not appear to affect playback in any of the players I have tried. I get this result with files produced with a lot simpler pipelines such as

`gst-launch-1.0 -ev udpsrc port=5600 ! application/x-rtp ! rtph264depay ! h264parse ! matroskamux ! filesink location="test.mkv"`

The warning appears in files that play perfectly, as well as those that do not. Notice that while the dts is not strictly increasing, it never decreases. Here is the language used by gstreamer documentation:
>The buffer DTS refers to the timestamp when the buffer should be decoded and is **usually** monotonically increasing. 

With monotonic dts aside, this command seems to reveal the problem:

`ffprobe -loglevel panic -of json -show_frames test.mp4`

The file created by the simple pipeline above produces this result for the first two frames, where the first dts time appears to be when everything starts running (not PLAYING, but when the caps spit out, so when caps are done negotiating or something):
>{
    "frames": [
        {
            "media_type": "video",
            "stream_index": 0,
            "key_frame": 1,
            "pkt_pts": 2082,
            "pkt_pts_time": "2.082000",
            "pkt_dts": 2082,
            "pkt_dts_time": "2.082000",
            "best_effort_timestamp": 2082,
            "best_effort_timestamp_time": "2.082000",
            "pkt_pos": "460",
            "pkt_size": "137126",
            "width": 1920,
            "height": 1080,
            "pix_fmt": "yuv420p",
            "sample_aspect_ratio": "1:1",
            "pict_type": "I",
            "coded_picture_number": 0,
            "display_picture_number": 0,
            "interlaced_frame": 0,
            "top_field_first": 0,
            "repeat_pict": 0
        },
        {
            "media_type": "video",
            "stream_index": 0,
            "key_frame": 0,
            "pkt_pts": 2109,
            "pkt_pts_time": "2.109000",
            "pkt_dts": 2109,
            "pkt_dts_time": "2.109000",
            "best_effort_timestamp": 2109,
            "best_effort_timestamp_time": "2.109000",
            "pkt_pos": "137594",
            "pkt_size": "72738",
            "width": 1920,
            "height": 1080,
            "pix_fmt": "yuv420p",
            "sample_aspect_ratio": "1:1",
            "pict_type": "P",
            "coded_picture_number": 1,
            "display_picture_number": 0,
            "interlaced_frame": 0,
            "top_field_first": 0,
            "repeat_pict": 0
        },

Files created by QGC produce results like this, where the first dts timestamp cooresponds to how long the stream has been running in QGC before the record button was pushed (~5 minutes in this case):
{
    "frames": [
[... 72 lines cut ...]
            "repeat_pict": 0
        },
        {
            "media_type": "video",
            "stream_index": 0,
            "key_frame": 0,
            "pkt_pts": 18,
            "pkt_pts_time": "0.018000",
            "pkt_dts": 18,
            "pkt_dts_time": "0.018000",
            "best_effort_timestamp": 18,
            "best_effort_timestamp_time": "0.018000",
            "pkt_pos": "110622",
            "pkt_size": "56765",
            "width": 1920,
            "height": 1080,
            "pix_fmt": "yuv420p",
            "sample_aspect_ratio": "1:1",
            "pict_type": "P",
            "coded_picture_number": 1,
            "display_picture_number": 0,
            "interlaced_frame": 0,
            "top_field_first": 0,
            "repeat_pict": 0
        },

NaterGator, 2018-01-16:
Sorry @jaxxzer, I should have been more specific when talking about clock distribution in gstreamer.  Adjusting the base time during recording is exactly what I had to do to fix this in another project. Good find!

DonLakeFlyer, 2018-06-18:
@jaxxzer Something that needs fixing here?

DonLakeFlyer, 2018-06-20:
Closing since nothing much going on here.,

closed 2018-06-20 by DonLakeFlyer

jaxxzer, 2018-06-28:
This should be resolved already. Sorry for the late reply.

PR #6034 (2018-04-02): Correct video file timestamping issues
Fix #5436.

This needs some testing and review, but appears to do the trick. I have one issue in that dropping frames or blocking on a pad probe in the recording branch, also appears to drop/block in the live stream branch. Though, from everything I have read, they should be independent. I must be missing something. The result is that when recording starts, the live stream stops until the next keyframe arrives, but only the recording branch should be waiting for the keyframe. @NaterGator @dogmaphobic, any clue?