mavlink/qgroundcontrol #5438: Please consider a proper QML widget interface, or widget examples.

AndKe, 2017-07-15:
I would love to see a widget example, that can read/write parameters, get/send mavlink, and know basic QGC status, like if mission is loaded, vehicle connected, parameters are loaded etc,

Having such framework, and access to all data from one place in the application, would make it simple to add specialized functionality, be it payload or vehicle related.
Also, it could, hopefully, bring more, and useful functionality without affecting the whole application.

DonLakeFlyer, 2017-07-15:
> I would love to see a widget example, that can read/write parameters, get/send mavlink

Hmm, there was one on the old web site:
* https://github.com/mavlink/qgroundcontrol/pull/1524
* http://qgroundcontrol.org/custom_command_qml_widgets

@hamishwillee Could you pull this across to the dev guide?

hamishwillee, 2017-07-17:
Yes, added docs in https://github.com/mavlink/qgc-dev-guide/issues/22

You'll need to update dialog to point to this once it is in.

closed 2024-06-01 by HTRamsey