mavlink/qgroundcontrol #5441: Resume Mission pops up sometimes after sliding Start Mission

Antiheavy, 2017-07-15:
Sometimes this happens:

- Slide Start Mission
- Vehicle switches to Mission and Armed
- Fixed Wing launch detection starts running (waiting for fwd acceleration to start motor)
- Resume Mission slider pops-up offering to "re-generate the mission from waypoint 1, takeoff, and continue the mission."

This doesn't seem right.  We are flying a fixed wing that uses "Launch Detection" in PX4.
[image]

DonLakeFlyer, 2017-07-15:
Did you do a change speed in Mission Start for this Plan? Wondering about the "1" item prior to takeoff.

I'm working on repro using HIL, but I think the problem may be that the firmware is changing the current mission item to 2 before it reports that it is flying. This is turn triggers the resume mission state.

@dagar Do you think my guess is possible?

Antiheavy, 2017-07-15:
> Did you do a change speed in Mission Start for this Plan?

I don't think so.  I did notice that the mission plan tacked on item (1) right near the Home point and made item (2) the Takeoff.   I'm not sure how this happened, maybe user error with clicks somehow?  Defintely didn't try to do anything different that we normally do.

dagar, 2017-07-15:
@DonLakeFlyer Yes, but that's what would happen if the first waypoint is a command. Does QGC insert speed changes now? If so you should put them after the takeoff (not just for this).

@Antiheavy Do you still have the plan? What's waypoint 1?

Antiheavy, 2017-07-15:
> Do you still have the plan?

Unfortunately did not save the mission plan.  I feel like we've seen this multiple times though so I will save it next time I see it.

DonLakeFlyer, 2017-07-15:
> I did notice that the mission plan tacked on item (1) right near the Home point and made item (2) the Takeoff.

Sorry. That's most likely the gimbal point down command.

DonLakeFlyer, 2017-07-15:
> but that's what would happen if the first waypoint is a command.

Hmm. That will be the case for a survey since when you add a survey it automatically adds a point gimbal down as the first item. Need to think about what to do about this.

DonLakeFlyer, 2017-07-15:
This bug was caused by the reversal of the order for Start Mission to:
1) Change flight mode to Mission
2) Arm

Still not quite sure how I can fix this given that new order.

dagar, 2017-07-16:
Can you expand on what's happening? Shouldn't all of these commands execute after takeoff? Once you're safely in the air you can point the gimbal, change speed, etc.

DonLakeFlyer, 2017-07-16:
> Shouldn't all of these commands execute after takeoff?

Theoretically yes, that would be better. But trying to make that happen is not easy to do. Something for the next version maybe.

> Can you expand on what's happening?

The signal for the current mission item changing comes through before the signal that the vehicle is flying comes through. This ends up being the same state triggers for when to show resume mission (mission item is not at the beginning or last and vehicle is not flying). This in turn cause resume mission to pop up for a second until the flying state goes true and causes it to go away. I need to change it to track whether the vehicle ever did make it into the air. And if that is the case then it is ok to show Resume Mission.

Antiheavy, 2017-07-16:
We saw the takeoff waypoint turning into waypoint 2 a bunch today at flight test.  I think it was happening once we uploaded the mission, the ordering would change and Takeoff would move to waypoint 2 and a new waypoint 1 was added next to Home.  We were using Survey patterns.  Also, sometimes waypoint 1 would be shown as active instead of Takeoff, which was confusing and disconcerting to the user.  However, "Start Mission" slider would set Takeoff as the new active waypoint (waypoint 2).  The aircraft flew the mission fine each time, flying to waypoint 2 (takeoff) and then continuing on.  Sometimes "Resume Mission" appeared and somtimes not.  We weren't able to fine a reliable pattern of behavior.

DonLakeFlyer, 2017-07-16:
> We saw the takeoff waypoint turning into waypoint 2 a bunch today at flight test.

This is because when you add a Survey to a mission a command to set the camera to photo mode is added to the start of the mission before the takeoff command.

PR #5448 (2017-07-16): Resume mission shows up at wrong times
Fixes #5441. Related to #5442
closed 2017-07-16 by DonLakeFlyer