mavlink/qgroundcontrol #5442: "Resume mission" problems with fixed wing

Antiheavy, 2017-07-15:
This may be fixed wing specific. When PX4 Fixed Wing does an RTL it switches the active waypoint to the "doLandStart" waypoint. Resume Mission seems to be picking up on that waypoint as the place to restart the waypoints. Activating resume mission slider then crashes QGC.  (I'm running windows version)
[image]

Also, after completing a normal landing sometimes Resume Mission pops up with waypoint 1 active instead of Start Mission popping up.
[image]

DonLakeFlyer, 2017-07-16:
The resume mission showing up at wrongs times is fixed now.

I can't seem to get the vehicle to go through the landing sequence when I do an RTL. It just flies immediately back to home and loiters there. @dagar Any ideas? Isn't FW supposed to use the landing pattern?

Antiheavy, 2017-07-16:
> Isn't FW supposed to use the landing pattern?

Using the landing pattern for RTL hasn't made it's way into mainline PX4 yet.  We've been test flying versions of this capability for a while now.  @dagar may have updates on future plans for this feature.

dagar, 2017-07-16:
@DonLakeFlyer it's in a PR here - https://github.com/PX4/Firmware/pull/7577

I've hesitated getting this into PX4 master because I'm not entirely sure if RTL triggering a mission landing is the best fit.

Questions
 - do we want to still preserve existing RTL behaviour? Maybe called RTH. Is it RTLand or RTLaunch?
 - what about the LAND mode? If you had a planned landing in place would you ever want to command a landing in place? That could still be preserved for failsafe triggers.
 - Smart RTL will hopefully make its way to PX4 soon (https://github.com/PX4/Firmware/issues/6464). How do these fit together?

Likely more info than you wanted, but would love to get some input.

Antiheavy, 2017-07-16:
Not related to the original topic of this thread, but I have opinions on your questions so I'm pleased to offer them:

> do we want to still preserve existing RTL behaviour? Maybe called RTH. Is it RTLand or RTLaunch?

The confusion between RTLand and RTLaunch is a pet peeve of mine - RTL is historically Return to Land in the world of small UAVs (maybe my opinion).  RTH is a very useful concept for VTOLs and is basically the standard expected behavior in the world of VTOLs now.  Updating the Home point upon takeoff is standard behavior for VTOL systems now-a-days and is essentially RTLaunch, but with the flexibility of moving Home during a flight if needed.

RTLand is the correct concept for fixed wing systems with a specific pre-planned landing location and pattern (or approach direction).  RTLand also works for VTOLs if users plan an explicit Land location and/or pattern.

I think the concept of RTLaunch creates more confusion than anything and RTH provides more utility anyway - RTLaunch should be purged.

I think I can explain this better in a table...  more to come.

> what about the LAND mode? If you had a planned landing in place would you ever want to command a landing in place? That could still be preserved for failsafe triggers.

Land mode (the concept of Land Now) is a basic expected capability for VTOL.  For Fixed Wing is it debatable if this capability should be presented to the operator.  It is important to have some sort of emergency landing capability in Fixed Wing, the best version of this is to be able to turn off propulsion and allow the controller to continue to navigate the waypoints as the aircraft glides.  In my opinion a dumb Land mode should be hidden behind the scenes for fixed wing and maybe used in fail safes, maybe not.

> Smart RTL will hopefully make its way to PX4 soon (PX4/Firmware#6464). How do these fit together?

cool idea.  i think this fits in well as an extension of the above RTH and RTL ideas.  It certainly does not replace RTH or RTL - it is more of an extension of the Min Return Altitude concept than anything.

DonLakeFlyer, 2017-07-16:
So Sentera firmware is running with this PR? If so, that complicates things a bunch.

The issue with respect the QGC and FW-RTL which re-uses the mission landing sequence is that it breaks the current Resume Mission QGC implementations since it moves the current mission item to the landing sequence. This in turn causes QGC to believe the full mission was run.

Antiheavy, 2017-07-16:
> So Sentera firmware is running with this PR? If so, that complicates things a bunch.
>
> The issue with respect the QGC and FW-RTL which re-uses the mission landing sequence is that it breaks the [...]

yep, to all.

dagar, 2017-07-16:
@DonLakeFlyer I'm not sure if I fully understand everything QGC is doing now, but the entire point is that you're using the landing planned within your mission. Longer term it might be nice if that landing was stored in a separate place, but you can imagine all the pieces involved.

Antiheavy, 2017-07-16:
I know this idea has potential problems as well, but maybe some sort of method for QGC to remember a mission item index from when the RTL event happened, then Resume Mission to that point?  Or maybe the previous mission index?  Or maybe create a new waypoint at the vehicle location when RTL was triggered and store that for use in regenerating the mission?

DonLakeFlyer, 2017-07-17:
> I know this idea has potential problems as well, but maybe some sort of method for QGC to remember a mission item index from when the RTL event happened, then Resume Mission to that point?

I'm going to have to figure out how to do something like that.

DonLakeFlyer, 2017-07-17:
@dagar Looking at the QGC code it already checks for not updating resume mission index if it is not in mission flight mode. Does the RTL landing thing switch the vehicle back to mission flight mode?

DonLakeFlyer, 2017-07-17:
I tried using your pull. Set the param to mission landing. But when I did an RTL the plane just seemed to wander off to nowhere. It did not change the flight mode to mission. So not sure how to handle this right now without something to test against. If the RTL does not change flight mode then resume mission should work as usual.

DonLakeFlyer, 2017-07-17:
I was going to wait on Stable 3.2.1 for this but I'm going to bail since it seems like there are too many questions around this to get it resolved quickly.

DonLakeFlyer, 2017-07-17:
@dagar I also tried building a gazebo version from the Sentera firmware repro so I could test with that but it give compile errors.

dagar, 2017-07-18:
It builds cleanly with the regular CI, what's the error? Ping me on slack tomorrow if you want to go through it.

DonLakeFlyer, 2017-07-29:
Ok I tested using https://github.com/PX4/Firmware/pull/7577. I still cannot test using Sentera firmware because it does not build with gazebo. From the QGC side of things resume mission worked correctly when the RTL went through the mission landing sequence to land. It remember the correct waypoint to rebuid from and rebuilt correctly.

What I did find was this:
* Use this Plan file as example: https://gist.github.com/DonLakeFlyer/2ac0be4e0db7c3d5f76882d724024ee1
* If you RTL when the current mission item is the one before the DO_LAND_START the plane wanders around aimlessly.

[image]

Antiheavy, 2017-07-30:
> the plane wanders around aimlessly

I believe we've had this happen to us during test flights a couple times.  Switching out of "return" mode back to "mission mode" stops the aimless wandering.  I'm not sure it was necessarily on the waypoint before DoLandStart for us though.  Is this repeatable?  I believe @dagar is investigating this bug and may or may-not have a proposed solution.

DonLakeFlyer, 2017-07-30:
> Is this repeatable in SIM?

100% percent repeatable for me using Gazebo.

DonLakeFlyer, 2017-12-29:
I beleve the resume mission generation problems with QGC are fixed now.

closed 2017-12-29 by DonLakeFlyer