mavlink/qgroundcontrol #5446: "Planned Home Position" confusion

dagar, 2017-07-16:
With PX4 the home position in planning continues to be a source of confusion. https://github.com/PX4/Firmware/issues/7623

PX4 doesn't have a planned home position and it always changes from you transition to flying. At a minimum could we remove "Planned"?

On the plan screen it needs to update with the HOME_POSITION from the vehicle, and probably shouldn't be user changeable.

jliphard, 2017-07-16:
Yes - that would be super. On the plan screen, call the window "Vehicle Home Position", show the actual values from the vehicle, and grey out the Lat/Lon/Alt numbers to visually communicate that they are un-editable since they are from the vehicle.

DonLakeFlyer, 2017-07-16:
> On the plan screen it needs to update with the HOME_POSITION from the vehicle

It already does that.

The reason for the "Planned Home Position" is so you can plan missions while not connected to a vehicle and still get valid stats and visuals for the mission. When planning offline if there was no planned home position you would not be able to show:
* Altitude displays
* Any stats which relate to mission distance
* Visual flight indicators which show the mission start/stopping at home

I do understand it is confusing. But I'm not sure getting rid of it entirely works either since you lose information.

DonLakeFlyer, 2017-07-16:
> On the plan screen, call the window "Vehicle Home Position", show the actual values from the vehicle, and grey out the Lat/Lon/Alt numbers to visually communicate that they are un-editable since they are from the vehicle.

You are running an old version of QGC. This is all quite different now. Only altitude is shown although it does not gray out when a vehicle is connected and sends a home position.

DonLakeFlyer, 2017-07-16:
[image]

jliphard, 2017-07-16:
Makes sense - could you have look at these screenshots and help me understand what they mean? Scenario - go to airfield, plug in autopilot, open up QGC. Home Altitude reported either as 122m (the correct value) or 0.00 or --.--. What does it mean when the altitude is 0.00 or --.--? No 3D GPS solution, other problem?

[image]

dagar, 2017-07-16:
> 
> > On the plan screen it needs to update with the HOME_POSITION from the vehicle
> 
> It already does that.

You're right when it changes on the vehicle and publishes a new HOME_POSITION, but there are a few situations where it changes. When downloading a mission from the vehicle or loading from file HOME resets to center. You're also able to drag the home point around.

DonLakeFlyer, 2017-07-17:
Yup. I'll look at this again in 3.2

Antiheavy, 2018-06-02:
Recent QGC daily builds have a different icon and name for the "Planned Home".  Simple solution.  I think it helps a fair amount with avoiding, at least some, confusion.
[image]

DonLakeFlyer, 2018-06-02:
I think this is about as good as it gets for now.

closed 2018-06-02 by DonLakeFlyer