mavlink/qgroundcontrol #5454: Ubuntu: "QGCGroupBox unavailable" in PowerComponent.qml

sanderux, 2017-07-17:
The power screen is doesnt render on ubuntu.
tested on 3.2 and latest master

The log shows "QGCGroupBox unavailable" in PowerComponent.qml, saving the log also fails.

[image]

[image]

sanderux, 2017-07-17:
PS: this is the appImage

DonLakeFlyer, 2017-07-17:
I just downloaded 3.2 AppImage onto Ubuntu 14.04 and ran without problem. Hmm?

kd0aij, 2017-07-17:
can't reproduce on Ubuntu 16.04 either

sanderux, 2017-07-17:
perhaps you guys have some other version of qt5quick ?

DonLakeFlyer, 2017-07-18:
I can reproduce on a clean 16.04

DonLakeFlyer, 2017-07-19:
@dagar @peterbarker Could any of the Linux folks out there take a look at this? I know nothing about AppImage.

peterbarker, 2017-07-19:
@DonLakeFlyer This is the first time I've heard the term "AppImage", sorry!

sanderux, 2017-07-19:
According to @dagar this is where to look: https://github.com/mavlink/qgroundcontrol/blob/master/QGCSetup.pri#L104

DonLakeFlyer, 2017-07-19:
Sorry @peterbarker I thought you had added the AppImage support. Got that wrong!

DonLakeFlyer, 2017-07-20:
@sanderux Can you try the latest daily build?

sanderux, 2017-07-20:
I tried latest daily (2017-07-19 18:23:39 -07.00) and still have the same issue.
i saw you added  libQt5QuickControls2.so.5
I think this should be libQt5QuickTemplates2.so.5

DonLakeFlyer, 2017-07-20:
Trying again. I have a feeling it needs both.

kd0aij, 2017-07-20:
confirmed: adding libQt5QuickTemplates2.so.5 fixes the PX4 "Power" menu page (not needed for ArduPilot)

DonLakeFlyer, 2017-07-20:
@sanderux Can you try as well? Want to make doubly sure before I announce 3.2 ready

sanderux, 2017-07-20:
I can do that in a few hours

sanderux, 2017-07-20:
Confirmed, works
[image]

DonLakeFlyer, 2017-07-20:
Thanks everyone

barchstien, 2017-08-09:
Hi guys
Thanks for the fix
Thought it still is failing when running the archive
The AppImage is just fine
So I guess it's only an update of the archive missing

DonLakeFlyer, 2017-08-09:
Ok, I guess that must be in a different place. I'll look.

DonLakeFlyer, 2017-08-12:
Doesn't the archive version require you to install Qt yourself?

DonLakeFlyer, 2017-08-17:
@dagar Isn't this the case?

dagar, 2017-10-08:
I believe it was able to run entirely from the contents of the archive at one point. I wouldn't mind dropping the archive and only having the appimage. Anything more than that should be a package.

DonLakeFlyer, 2017-10-10:
I would like to drop the archive as well. Too many things to keep up to date. Only one thing to keep running would be great.

DonLakeFlyer, 2017-12-29:
I'm going to drop the archive from the docs for now and see if anyone complains.

closed 2017-12-29 by DonLakeFlyer