mavlink/qgroundcontrol #5467: build qgroundcontrol3.1.3 in ubuntu 14.04 error

pyh0603, 2017-07-18:
after me follow the teps as :https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html#ubuntu-linux。then i run ./qgroundcontrol-start.sh.  diplay some error as list.who can tell me how to solve it???

pyh@pyh:~/dev/qgc/qgroundcontrol$ ./qgroundcontrol-start.sh 
Settings location "/home/pyh/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
MAVLinkLogManagerLog: MAVLink logs directory: "/home/pyh/Documents/QGroundControl/Logs"
Map Cache in: "/home/pyh/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: 0.03518518518518519
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
qml: 34.5 38
qml: _activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()
VMware: vmw_ioctl_command error Invalid argument.
Aborted (core dumped)

pyh0603, 2017-07-18:
one more,i have  install gstreamer packages as follow:https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming.but it still does work!

DonLakeFlyer, 2017-07-18:
GitHub Issues are for bugs only. For developer support, use the various channels shown here: http://qgroundcontrol.com/#resources

closed 2017-07-18 by DonLakeFlyer