mavlink/qgroundcontrol #5478: Show airspeed as disabled, not 'setup required'

sanderux, 2017-07-21:
[image]

[image]

dagar, 2017-07-21:
Can we discuss a way to handle this via a mavlink api? The requirements for which sensors need to be available and calibrated based on vehicle type and configuration are much easier to handle on the firmware side.

DonLakeFlyer, 2017-07-22:
> The requirements for which sensors need to be available and calibrated based on vehicle type and configuration are much easier to handle on the firmware side.

That's for sure. If you have something to propose, let's do the discussion in mavlink github since we ened ArduPilot folks involved as well.

dagar, 2017-07-30:
@DonLakeFlyer what about doing it via SYS_STATUS?
Either using the existing fields, or adding new ones to indicate which are required and calibrated. Do you need to know about every sensor, or just per sensor class?

DonLakeFlyer, 2017-07-30:
Every sensor since there are red/green indicators on compass/gyro/accel as well.

dagar, 2017-07-30:
I mean one bit for each
 - compass
 - gyro
 - accel
 - airspeed

or
 - compass1, compass2, compass3, compass4
 - gyro1, gyro2, gyro3 
 - accel1, accel2, accel3
 - airspeed

Given you can only calibrate them in bulk I think that answers the question.

DonLakeFlyer, 2017-07-30:
> I mean one bit for each
> 
> compass
> gyro
> accel
> airspeed

Yup, that's what I would need.

dagar, 2017-07-30:
So do you think that's an appropriate interpretation of onboard_control_sensors_{present, enabled, health}, or do we need new fields?

onboard_control_sensors_present - all sensors found active on the system
onboard_control_sensors_enabled - autopilot sets enabled if that sensor is needed for this configuration
onboard_control_sensors_health - sensor wouldn't be marked healthy if calibrated, but then we couldn't distinguish actual unhealthy checks later.

http://mavlink.org/messages/common#SYS_STATUS

If it weren't for the Mavlink v1/v2 problems I'd simply want to add onboard_control_sensors_calibrated.

DonLakeFlyer, 2018-01-03:
Correctly respects FW_ARSP_MODE now. The whole concept of how this is done in QGC still needs revisit tough.

closed 2018-01-03 by DonLakeFlyer