mavlink/qgroundcontrol #5479: Problem with Qgroundcontrol build process with Qt in windows

alireza787b, 2017-07-21:
I successfully compiled Qground in macOs with Qt. I am looking for a way to build my modified project for windows. first I tried cross compiling in Mac for windows but couldn't and I face with tons of errors.
I installed Qt 5.7.1 on my windows and opened my modified Qt project . in configuration / Kits I see that MSVC12 x86 is listed but when  I click on that I see that compiler drop down menu is disabled and no compiler is selected. I tried creating a new kit and manually select MSVC 12  and Qt version of 5.7.1. it compiles but I faced with hundreds of syntax errors... am I missing sth? I have already installed VS 2015 , 2013 on PC as well.

DonLakeFlyer, 2017-07-21:
GitHub issues are for bugs. For developer issues please use one of the appropirate discussion forums.

closed 2017-07-21 by DonLakeFlyer

alireza787b, 2017-07-22:
so where should. I ask my question?

DonLakeFlyer, 2017-07-22:
http://qgroundcontrol.com/#resources