mavlink/qgroundcontrol #5497: Not possible to load airframe paraameter for newest ArduCopter

msadowski, 2017-07-25:
I just downloaded QGroundControl v3.2.2 and flashed it with the latest ArduCopter V3.5.0.
There are two problems I found in the airframe tab:
1. Pressing Load common parameters doesn't respect Frame Class and Type selected from the dropdown menu, always the same parameters are shown.
2. Some links to parameter files seems to be outdated (for example 3DR QUAD X4 RTF) returns 

> Param file github json download failed: Error: File Not Found

DonLakeFlyer, 2017-07-25:
> Pressing Load common parameters doesn't respect Frame Class and Type selected from the dropdown menu, always the same parameters are shown.

This is working correctly, if not super useful at this point. Filtering by frame/class is NYI. Will change in 3.3

> Some links to parameter files seems to be outdated (for example 3DR QUAD X4 RTF) returns

Will take a look at this for 3.2

DonLakeFlyer, 2017-08-01:
Looks like the ardupilot folks deleted most of the .param files for airframe setup. Trying to find out why.

DonLakeFlyer, 2017-08-03:
Talked with ardupilot these param setup files are no longer supported and have been removed. So unfortunately I"m going to have to remove this feature from QGC.

DonLakeFlyer, 2017-12-30:
Removed

closed 2017-12-30 by DonLakeFlyer