mavlink/qgroundcontrol #5511: Android Beta Icon

dagar, 2017-07-30:
In the Play store the QGroundControl and QGC (Daily Test Build) have slightly different icons, but on the active device they're using the same icon.

[image]

We need an a better beta/test icon generated for android in all sizes, then build scripting to change the icon here. https://github.com/mavlink/qgroundcontrol/blob/master/android/AndroidManifest.xml#L3
https://github.com/mavlink/qgroundcontrol/blob/master/tools/update_android_version.sh#L16

hamishwillee, 2017-07-30:
The UX team at Intel are going to love the opportunity to help with this. BUT if we want help we need to provide very clear direction. So, list of what you want ...

dagar, 2017-07-31:
Something similar to what Google did with Android Chrome?

[image] https://play.google.com/store/search?q=chrome

hamishwillee, 2017-08-01:
Also, are we happy with the current iconography, or tempted by elements of the proposed DC branding: https://www.youtube.com/watch?v=i4kblPyLdBw&feature=youtu.be

W.r.t. the above @dagar I like that, but this exact approach might cut the bottom off the Q. All we want is something that is recognizably QGroundControl and says "Beta" right?

anassinator, 2017-08-01:
@dagar, @hamishwillee:

From my experience dogfooding, I've seen two ways of distinguishing the beta/nightly builds from prod:
- Adding a banner as Chrome does above
- Changing the colors or background somehow to make it look like a WIP or different enough from prod such that they can't be confused

What do you guys think of changing the background instead? This is a quick and rough example of what I mean:

[image]

Original for comparison:

[image]

On a side note, I'd love to see a more DC-inspired icon.

dagar, 2017-08-01:
I'm not sure if the default QGC coloring is sufficiently recognizable at this point. Can you still slap a beta label across it?

DonLakeFlyer, 2017-08-01:
like the graph paper style thing. But agree that some sort of Beta label is likely still needed.

NaterGator, 2017-08-01:
I don't have an artistic bone in my body or I would mock this up, but could you stylize the "Q" portion of the beta logo to look like a sketch / draft of the letter's geometry (roughly this premise: http://imgur.com/a/c1Kiu)

hamishwillee, 2017-08-02:
@NaterGator Looks cool - but IMO a bit far from the original :-) Lets get the UXWG team to do the actual work of producing the logo. What we need to do is provide them with ideas on what we think might work/we could live with. 

I think we all want "Beta" on the logo. We all want this to remain clearly "QGroundControl". We like the graph paper thing, but open to other ideas. Reasonable feedback to give them?

anassinator, 2017-08-02:
@NaterGator I think the sketch could be a neat idea, but don't have the time to whip that up right now.
@hamishwillee I think he meant we could have the entire icon be the same as the original but to look completely hand-drawn on a blueprint or something instead. Correct me if I'm wrong.

I am quickly putting the following up simply for a better idea of what this could look like.

Blueprint, bottom label:
[image]

Blueprint, corner label:
[image]

Original, bottom label:
[image]

Original, corner label:
[image]

The corner label can obviously be on any corner. I also didn't put any thought into the colors, fonts, sizing, background and etc. I'm not a designer and I'm certain the UXWG team can do a better job; I'm just trying to help out so that you may have something more concrete to give.

Also, this probably goes without saying, but I think we should do the same for the desktop apps and iOS (not sure if the TestFlight is still a thing).

NaterGator, 2017-08-02:
Yeah, I meant basically composite what looks like sketching the geometry of the Q in to give it a "work in progress" feel. To be honest 1 and 2 look pretty great and are a quick glance to realize this is not the stable release branch.

hamishwillee, 2017-08-02:
Yes, they (https://github.com/anassinator) all look really good to me. I slightly prefer the graph background, and the "corner" orientation. If we're all basically OK, I can forward this to UXWG?

>  think we should do the same for the desktop apps and iOS (not sure if the TestFlight is still a thing).

So I was assuming we'd use same Beta icon for all platforms. Is that what you're saying?

FYI I did understand what @NaterGator was suggesting. I don't hate the idea. To me though it felt more like something to capture a prototype :-) than a beta.

anassinator, 2017-08-03:
@hamishwillee 

Yeah it should be the same design, but each platform has different sizes and formats requirements. For example, the iOS icon must not be rounded as the OS adds that itself.

Also, I would suggest either reworking the rounding radius of the icon for each platform as it seems too harsh right now. I haven't been following Android very much lately but I think I've seen a lot of completely circular (like chrome above) and completely square icons. And for example Mac or has a lot more circular icons as well. So, maybe we should investigate shapes that better suite each platform. This can also affect whether corner or bottom makes more sense.

And ok was just making sure we're on the same page 🙂

Finally, moving forward with UXWG isn't my call, but I'm OK with that 😛

DonLakeFlyer, 2017-08-03:
I'd rather not have different branding on each platform. Also I'd rather not change the existing branding (other than a beta version) at this point. You bounce around too much and people don't know what's what.

hamishwillee, 2017-08-03:
@DonLakeFlyer Thanks for that Don. I'll see whether UXWG care to comment. If they don't come back with clear guidance or a plan for clear guidance in a week, I personally would be happy with us all to make an executive decision :-) Perhaps on option 2 [above](https://github.com/mavlink/qgroundcontrol/issues/5511#issuecomment-319661447)

DonLakeFlyer, 2017-08-03:
I think this one is fine:
[image]

hamishwillee, 2017-08-03:
Me to. Just sent an email to UXWG for comment, listing that as my preference and asking for comment/professional versions in the right sizes. Lets give them a few days to respond - OK?

anassinator, 2017-08-03:
@hamishwillee, @DonLakeFlyer:

That's good by me.

For reference, since I didn't see these comments in time, this is what I meant about a circular icon:

[image]

iOS would need to remain a square, but IMO this isn't changing the branding.

DonLakeFlyer, 2017-08-03:
> but IMO this isn't changing the branding.

The square with rounded edges are definitely part of the banding. It's used in many places.

anassinator, 2017-08-03:
Ah I see what you mean. Fair enough 👍

DonLakeFlyer, 2017-08-03:
Change title since we want to do this for all OS.

fabricionovak, 2017-08-03:
The blueprint background and Beta tag will do the job. My only concern is about legibility in different sizes. I've made some tweaks and tests.

[image]

[image]

[image]

[image]

anassinator, 2017-08-03:
That looks nice.
Yeah the beta label might be hard to read on lower resolution devices. That's why a noticeably different background is common and would probably be the best indicator IMO. We can still have the label as well.

fabricionovak, 2017-08-03:
@anassinator I'd avoid changing the background color. The color combination is part of the branding guidelines. The beta label is small, but it is pretty readable in my opinion. Here is a picture of my phone right now (not a mockup like the previous one).

[image]

anassinator, 2017-08-03:
@fabricionovak:
Oh I thought you were worried about lower DPI devices where the label might be much harder to make out.
As for the branding issue, I understand the concern, and it's up to you guys for sure. I prefer the one you posted anyways :)

fabricionovak, 2017-08-03:
These lower DPI devices will not display the text as sharp as the high-end ones, but it will be still readable.

hamishwillee, 2017-08-03:
OK, @fabricionovak I think we'll all be happy with this version you have proposed. Can we have versions suitable for desktop, android and iOS? [Do you need list of specific sizes/formats?]

fabricionovak, 2017-08-03:
@hamishwillee a list would be helpful. :)

hamishwillee, 2017-08-04:
@fabricionovak The android icon requirements are here: https://developer.android.com/guide/practices/ui_guidelines/icon_design_launcher.html#size
(note the need for the android play store icon in the following section).

iOS app icon sizes are here: https://developer.apple.com/ios/human-interface-guidelines/graphics/app-icon/

@DonLakeFlyer @dagar Is that correct set? What sizes are needed for desktop?

hamishwillee, 2017-08-08:
@fabricionovak I know you're busy. Can you please ETA some of these icons?

fabricionovak, 2017-08-09:
@hamishwillee I'll work on this today.

hamishwillee, 2017-08-09:
Thanks @fabricionovak  !

fabricionovak, 2017-08-09:
Android icons in PNG. Still working on the iOS versions

[image]
[image]
[image]
[image]
[image]
[image]

fabricionovak, 2017-08-09:
PSDs for android

[ic_qgc_beta_android.zip](https://github.com/mavlink/qgroundcontrol/files/1213061/ic_qgc_beta_android.zip)

hamishwillee, 2017-08-09:
Thank Fabricio! @dagar @DonLakeFlyer  What else do you need/can you take it from here?

DonLakeFlyer, 2017-08-10:
I'll get going on this. I'll yell if I need antyhing.

fabricionovak, 2017-08-14:
PNG files for iOS devices
[image]
[image]
[image]
[image]
[image]

fabricionovak, 2017-08-14:
iOS icons in PSD
[ic_qgc_beta_iOS.zip](https://github.com/mavlink/qgroundcontrol/files/1223560/ic_qgc_beta_iOS.zip)

fabricionovak, 2017-08-14:
Android icon update in XXXHDPI. Please, replace the previous files (ic_qgc_beta_android_XXXHDPI.psd and ic_qgc_beta_android_XXXHDPI.png) for these.

[image]
[ic_qgc_beta_android_XXXHDPI.zip](https://github.com/mavlink/qgroundcontrol/files/1223566/ic_qgc_beta_android_XXXHDPI.zip)

DonLakeFlyer, 2017-08-17:
Won't be closed until their use is implemented.

DonLakeFlyer, 2018-01-07:
I started hooking these up, but they won't go in until 3.3 is out the door. Then dev/daily builds will cut over to these.

Prasad28-97, 2021-06-16:
Hii Guys 
I want to change the default icon of QGC in my project. Can you tell me exactly where i need to make changes?

Thanks

closed 2026-10-04 by github-actions[bot]