mavlink/qgroundcontrol #5519: Problems flashing firmware with v3.2.2

larsks, 2017-08-02:
I am trying to use qgroundcontrol v3.2.2 to flash a pixhawk clone ([this](http://ardupilot.org/copter/docs/common-pixhawk-overview.html) kind of pixhawk).  The update process gets as far as "Programming new version..." and then gets stuck.

I am running on Ubuntu 17.04.  This is not a permissions or general serial port problem: qgroundcontrol, mavproxy, and apmplanner are all able to connect successfully to the pixhawk.  I was able to use apmplanner to install the firmware upgrade.

As requested by @DonLakeFlyer, here is a screenshot of the firmware upgrade log:

[image]

larsks, 2017-10-31:
@DonLakeFlyer This problem still exists with the latest daily build. Do you have any thoughts on a solution?  Is there any additional information I can provide?

DonLakeFlyer, 2017-10-31:
Can you turn on FirmwareUpgradeLog as shown here: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html. Save the log to a file. And paste or attach here.

Changliu52, 2017-11-07:
same problem here for v3.2.5 on ubuntu 16.04.
Console logging here:
```
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:259 - "Settings location "/home/chang/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:99 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/Vehicle/MAVLinkLogManager.cc:359 - "MAVLink logs directory: "/home/chang/Documents/QGroundControl/Logs""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QtLocationPlugin/QGCMapEngine.cpp:215 - "Map Cache in: "/home/chang/.cache/QGCMapCache300" / "qgcMapCache.db""
[D] at qrc:///qml/QGroundControl/ScreenTools/ScreenTools.qml:117 - "0.057291666666666664"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:130 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:130 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/Controls/SliderSwitch.qml:74 - "34.5 44"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:130 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/MAVLinkProtocol.cc:413 - "Orphaned log file count 0"
[E] at :0 - "QDBusConnection: name 'org.freedesktop.Geoclue.Master' had owner '' but we thought it was ':1.72'"
```

larsks, 2017-11-07:
Here are two console logs, one from the current release and one from the latest daily:

- [release](http://chunk.io/larsks/8d7bcf658f344ecf80c936808b01027d)
- [daily](http://chunk.io/larsks/3eebacfb7ac24c7b94d8aa9c50f4b387)

closed 2017-11-12 by DonLakeFlyer

glynch123, 2018-02-04:
Hi. I am having the exact problem as larsks. How did you resolve the issue?

DonLakeFlyer, 2018-02-05:
Are you using a daily build?

glynch123, 2018-02-06:
It worked with daily build! Thank you!