mavlink/qgroundcontrol #5524: Support terrain elevation data for mission planning

Antiheavy, 2017-08-03:
This is a big can-of-worms topic, but here are some of the things involved from my experience:

- Support free on-line and off-line terrain elevation data.

- Warning messages and indicators if terrain elevation data option is enabled, but on/off-line data is not available for the planned area.

- Default mission altitudes are relative to Home, except when a "min height above terrain" setting is reached, then apply a positive offset to preserve min height above terrain with appropriate warning/info message.

- Checkbox option for all planned waypoints to be height above terrain instead of height above Home (defaulted to OFF). When toggled globally apply changes to already planned waypoints.

- Perform Line-of-sight calculation from planned waypoints and/or planned flight paths to GCS location (use Home point if GCS location not known). Checkbox option to prevent placing of waypoints that violate line-of-sight (default to ON). Use appropriate buffers in geometry calculation, e.g. 1m above home point, ignore 10 m radius around home point, use 10 m buffer above terrain. Display appropriate warning/info messages.

- Issue warning about safe return home altitude if: terrain within planned area >"Safe return altitude" - "min height above terrain". I've seen discussions about future firmware support for advanced safe RTL that might make this more complicated, but until that exists...

- toggleable map-layer that shows line-of-sight boundaries at various altitudes above GCS/Home location (like a Topo map overlay).

Other related topics:
https://github.com/mavlink/qgroundcontrol/pull/5513
https://github.com/mavlink/qgroundcontrol/issues/4007
https://github.com/mavlink/qgroundcontrol/issues/2927
https://github.com/mavlink/qgroundcontrol/issues/1643

birchera, 2017-08-03:
Nice to see all the progress here! I would like to add one thought as well:
* When flying with a certain automated policy with respect to altitude over ground (e.g. fixed) feasibility is vehicle dynamics dependent. E.g. fixedwings only rise or sink at a certain rate.

Antiheavy, 2018-03-13:
see related terrain elevation discussion here: https://github.com/mavlink/qgroundcontrol/issues/6204

closed 2018-06-20 by DonLakeFlyer